COVESA/vehicle_signal_specification #252: Tree pruning

UlfBj, 2021-03-15:
I would like to propose a solution to two issues,
1. to create asymmetric matrices using the instantiation mechanism
2. to remove nodes using the VSS layering mechanism

The idea is to introduce a new key-value property (metadata) named "prune", that would have a value in the format of either a single string or an array of strings. This value would consist one or more relative paths, representing absolute paths in combination with the path to the node in which it is found. The "pruning", i. e. removal of nodes would then be applied by the tool for the node addressed by the absolute path, and any subtree below this path. 
An example of 1. is shown below where the instantiation will lead to a tree representing Pos1, and Pos3 seats in Row1, Pos1, Pos2, and Pos3 seats in Row2, and a Pos2 seat in Row3.
```
- Seat:
    type: branch
    instances:
    - Row[1,3]
    - Pos[1,3]
    prune: [Row1.Pos2, Row3.Pos1, Row3.Pos3]
    description: All seats.
```
An example of 2. could be the following that would remove the entire OBD branch, assuming it is part of the YAML tree. 
```
- Vehicle:
  type: branch
  prune: OBD
  description: High-level vehicle data.
```
The "prune" property would typically be added in the deployment phase using the layering mechanism, and then it would be the responsibility of the tool to prune its output accordingly.

danielwilms, 2021-03-30:
First define basic description of layering. Good possible solution for deletion.

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch