COVESA/vehicle_signal_specification #258: Signals of heavy-duty vehicle

iabtv, 2021-03-25:
Hello,

I'm working on a private branch with Signals of heavy-duty vehicle,  most of them refer to SAE J1939-71/73. Is there any plan add such signals to the standard branch? 

e.x. 
```
- Engine:
  type: branch
  description: Engine signals

- Engine.TorqueMode:
  datatype: string
  type: sensor
  enum: ['not_request', 'accelerator_pedal', 'cruise_control', 'pto_governor', 'road_speed_governor', 'asr_control', 'transmission_control',
  'abs_control', 'torque_limiting', 'high_speed_governor', 'braking_system', 'remote_accelerator', 'not_define', 'other', 'not_available']
  description: Reference SAE J1939-71 SPN899. State signal which indicates which engine torque mode is currently generating, limiting, or controlling the torque.

#
# Current percentage Torque
#
- Engine.ActualPercentTorque:
  datatype: uint8
  type: sensor
  unit: percent
  min: 0
  max: 125
  description: Reference SAE J1939-71 PGN61444/SPN513. The calculated output torque of the engine.

- Engine.IsCompressionReleaseActive:
  datatype: boolean
  type: actuator
  deescription: Indicates if the engine compression release control system is enalbed. True = Enabled. False = Disabled.

```

gunnarx, 2021-04-01:
Hello @iabtv 

First the answer to your question.  Note I am trying my best to represent the view of many stakeholders according to previous group discussions about the VSS.

> I'm working on a private branch with Signals of heavy-duty vehicle, most of them refer to SAE J1939-71/73. Is there any plan add such signals to the standard branch?

Many agree that we are aiming for a data model that is applicable in both personal vehicles and heavy vehicles.  This means the VSS standard catalog could include signals that are unique to heavy vehicles, and/or we make a combination of more than one standard catalogs, or use appropriate specific branches in the VSS tree for anything that is unique to heavy vehicles.  Some companies may choose to represent the J1939 in a VSS-formatted tree in order to use VSS-compatible software, but again an alternative is to just fill the current VSS catalog with information that is needed, with no particular regard to J1939.  

I think the first step is to identify what is needed, make comparisons, and discuss options.

Some thoughts (we discussed this in the weekly VSS development meeting).

1. For your own development, do you need the exact J1939 list of data items, or do you need only to add some information that you find missing in VSS?

2. The group generally prefers to put overlapping things  in its "natural" place in the existing VSS tree instead of creating a J1939 specific tree, but I think the final decisions depend on what the options are.  Some analysis of the content and overlap is needed.

3.  If we define new data items in the "natural" position in the VSS tree rather than according to the J1939 standard, some companies may still benefit from having a mapping/translation table.  It could be used by anyone who is used to J1939, or has legacy software designed specifically for J1939, or other reasons.  These scenarios are worth discussing.

Would you consider joining the weekly VSS development meeting to discuss this in more detail?

iabtv, 2021-04-07:
Hi @gunnarx 

Thanks for your detailed reply and invitation. I'm very excited to contribute.

Please let me share more information according to your thoughts:

> 1. For your own development, do you need the exact J1939 list of data items, or do you need only to add some information that you find missing in VSS?

For current project, I only need to define a subset signals of J1939 and specific engine controller unit to VSS tree. They will be added to the private branch only if they are missing from standard one. I also try my best to deliberate how to organize them and keep them avaliable for future modification. 

> 2. The group generally prefers to put overlapping things in its "natural" place in the existing VSS tree instead of creating a J1939 specific tree, but I think the final decisions depend on what the options are. Some analysis of the content and overlap is needed.

Totally agree. If it won't boring you, I'd like share this project-oriented case after I complete definition of the private brach in a few days.

> 3. If we define new data items in the "natural" position in the VSS tree rather than according to the J1939 standard, some companies may still benefit from having a mapping/translation table. It [...]

Mapping/Translation table is fine. The big issue for me is the serialization solution with google protocol buffer which has position aware elements. That's why I take advantage from J1939 PGN to my private branch.

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.