COVESA/vehicle_signal_specification #265: Redefinition of instances - how it is intended to work?

erikbosch, 2021-04-12:
How is redefinition of instances supposed to work? We have in VSS in quite a lot of places default instance declarations, like e.g. having a default of 2 axles. But what if you instead want to have 4 axis? You cannot just add the rows below to an arbitrary file (after the original declaration) like:

```
#overwrite, use 4 axis instead of 2
- Axle:
  instances:
    - Row[1,4]
  type: branch
  description: Axle signals
```
... because then you will (at least for CSV) loose the member declarations (if a branch is redefined members previously defined for that branch are lost), and the instances will not be replaced but rather appended as a new dimension. I have not looked at the other tools.

Is this a defect or a feature?

```
Vehicle.Chassis.Axle,branch,,,false,,,,Axle signals,,0a3ebde7efa85c04ac6c29b5676fec5d,
Vehicle.Chassis.Axle.Row1,branch,,,false,,,,Axle signals,,d7e93a94af0752aaab36819f6be4f67a,
Vehicle.Chassis.Axle.Row1.Row1,branch,,,false,,,,Axle signals,,640788ca644d5c2b8323ce549098eb0d,
Vehicle.Chassis.Axle.Row1.Row2,branch,,,false,,,,Axle signals,,b65aa79ec20a591e8a5ed58300ab4d08,
Vehicle.Chassis.Axle.Row1.Row3,branch,,,false,,,,Axle signals,,b63328b51063528e83faa710cad47d80,
Vehicle.Chassis.Axle.Row1.Row4,branch,,,false,,,,Axle signals,,c339826203d85b9597229ac0447606f8,
Vehicle.Chassis.Axle.Row1Count,attribute,uint8,,false,,,,Number of axles on the vehicle,,f5d7402c65bd5d29b8167fe480c76042,
Vehicle.Chassis.Axle.Row2,branch,,,false,,,,Axle signals,,8ef77768446659b6b5020a06c7b23c8b,
Vehicle.Chassis.Axle.Row2.Row1,branch,,,false,,,,Axle signals,,3565bae8d717580698d92cf0c963dbd8,
Vehicle.Chassis.Axle.Row2.Row2,branch,,,false,,,,Axle signals,,0801d2fef0095bd7a5271a5f891b75e3,
Vehicle.Chassis.Axle.Row2.Row3,branch,,,false,,,,Axle signals,,2456c02da3be5e71a6fbed9c5b11be7d,
Vehicle.Chassis.Axle.Row2.Row4,branch,,,false,,,,Axle signals,,d6280501e4fc5ab9bbcd881d1b83989e,
```

danielwilms, 2021-04-20:
@erikbosch please add a ticket to tools

danielwilms, 2021-06-01:
@erikbosch, this can be closed, right?

closed 2021-06-01 by erikbosch