COVESA/vehicle_signal_specification #268: Splitting (meta)-model from data catalogue: Step by step

SebastianSchildt, 2021-04-20:
Practitioners guide to the metaphysics discussions that took place in our meetings/Slack

Situation: There is a rough consensus, that some discussion, especially layering and "scope" of VSS can benefit from a clearer separation between the "model" (i.e. the rules how to create a syntactically valid VSS tree) and the "catalogue" (the concrete names and datatypes defined), where the model would also be applicable to other models or can be used to model different automotive subdomains

We need to do the following

1.  We need to keep the name VSS for.. VSS, our current catalogue as it is widely known
2. We need to change the name for the data model. Let's _not_ discuss this here. For the sake of this list I use SignalDataModel (SDM)
3. Rename VSS-tools to SDM-tools (or SDM)
4. Split the documentation https://genivi.github.io/vehicle_signal_specification/introduction/   there may be finer points but roughly, here is a list what stays in VSS and what moves to SDM (i.e. having two websites)
  - Introduction: Largely VSS (may need a fre new words for SDM)
  - Ruleset->Basic Rules: SDM
  - Ruleset->Branch Entry: SDM
  - Ruleset->DataEntry->DataTypes: SDM
  - Ruleset->DataEntry->DataUNitTypes: VSS (some generic comments in SDM)
  - Ruleset->DataEntry->Sensors&Actuators: SDM
  - Ruleset->DataEntry->Enumerated Sensors & Actuators : SDM (side note, is it even use for anything except strings in VSS?)
  - Ruleset->DataEntry->Attributes: SDM
  - Ruleset->Instances: SDM
  - RuleSet->Includes: SDM
  - RuleSet->PrivateBranch: VSS
  - Tools: SDM
5. "Makefile" needs to live in VSS (SDM might have a seperate one, for testing purposes), but one for both as we try now would not be "logical" anymore

Once that is done: Nothing changed at all for users

Layering/Composition would be described in terms of SDM

Any concrete example for e.g. modelling access rights would be different SDM model (might of course link as "recommended")

Any extension e.g. full J1939 support, airplanes, bikes might be different SDM modles or layers "designed" to be applied on top of VSS

erikbosch, 2021-04-21:
For unit types (like kPa, mpg, g/km, ... ) it can be discussed if it should belong to VSS or SDM. Standardized types on meta-model level would simplify handling of e.g. automatic unit conversion in tools/services for various catalogs. If we decide to put the unit types to VSS then they must be fed to SDM tools as e.g. an input parameter.

danielwilms, 2021-04-22:
> For unit types (like kPa, mpg, g/km, ... ) it can be discussed if it should belong to VSS or SDM. 

I would say they belong in SDM, as there should be an agreement in all derived models, especially if you want to link them

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch