COVESA/vehicle_signal_specification #270: CSV for VSS incorrectly formatted

nickrbb, 2021-04-26:
Hi,

I noticed that the CSV format of the VSS has the common issue where fields are incorrectly delineated due to commas appearing in Description and Enum field data. The CSV file format allows overcoming this issue by encasing all CSV field data within double quotes (see IETF RFC 4180 for more information). To this end, may I recommend that in module vspec2csv.py, function "print_csv_content", be updated to ensure the field content of the VSS CSV file is delimited by double quotes?

In my testing, just updating lines 64 and 65 (or thereabouts) of this module of code to the following had the desired effect:
`f"\"{tree_node.qualified_name('.')}\",\"{tree_node.type.value}\",\"{data_type_str}\",\"{tree_node.deprecation}\",\"true\",\"{unit_str}\",\"{tree_node.min}\",\"{tree_node.max}\",\"{tree_node.description}\",\"{tree_node.enum}\",\"{tree_node.uuid}\",\"{instance}\"\n")`
and
`f"\"{tree_node.qualified_name('.')}\",\"{tree_node.type.value}\",\"{data_type_str}\",\"{tree_node.deprecation}\",\"false\",\"{unit_str}\",\"{tree_node.min}\",\"{tree_node.max}\",\"{tree_node.description}\",\"{tree_node.enum}\",\"{tree_node.uuid}\"""\n")`

This then produces a proper delineation of field data in the VSS CSV file, making the viewing of VSS in the likes of Microsoft Excel much easier and more cleaner to look at due to correct delineation of the fields.

Thanks,
Nick

gunnarx, 2021-04-26:
@nickrbb Great that you have this solved. Can you send in a code change proposal (Pull Request)?

erikbosch, 2021-04-27:
Supporting comma is great, but I think that we also need to discuss the valid character set for various fields, describe it in the documentation and preferably also let the tooling enforce it. As an example, it is fully possible today to have the description below and generate csv from it without complaints, but the csv generated is not that useful. Should we for example allow double quotes in description (and let tools like vspec2csv add escapes as needed), or should we rather avoid the problem by saying that double quotes are not allowed and preferably ignore them and generate an error? 

(but this can be handled separately from the change proposed above)

```
- My,Si"gnal:
  datatype: string
  type: attribute
  description: Do you want some "Smörgåstårta", it is great!
```

danielwilms, 2021-04-27:
Things to check:
- Documentation
- Tooling 
- Test cases
- YAML schema (?)

SebastianSchildt, 2021-04-27:
Can this help https://json-schema-everywhere.github.io/yaml

SebastianSchildt, 2021-04-27:
The 120% solution: https://www.globalnerdy.com/2014/06/03/swift-fun-fact-1-you-can-use-emoji-characters-in-variable-constant-function-and-class-names/ 😁

erikbosch, 2021-04-28:
Assuming that we only allow "unexpected" double quotes in the description field, we could add something like below to `print_csv_content` to escape double quotes inside description.

`descr_str_escaped = tree_node.description.replace('"','""')`

nickrbb, 2021-04-28:
Thanks for all of the above replies. I think we should scope this particular issue for now to just enabling correct CSV syntax to be produced, which means allowing for the use of commas as well as double quotes (and even emoji's!) in any field. Another, separate exercise to validate the contents of each field based on a set of allowable chars would definitely be useful (e.g. prevent unit8[] appearing as a data type, which I see it does for Vehicle.Cabin.SeatPosCount) but that should be done as part of a YAML file check exercise at a point before any output files are generated e.g. CSV, JSON, etc. I haven't looked into the latter yet, so don't know what checks are already done and the handling of such issues.

Going back to the CSV generation, perhaps it would be better to have a function specifically for producing correctly formatted CSV fields. So a function that can accept an unbounded number of string variables, and which then loops through each one to produce a CSV data line. Within the loop we can then ensure such things as double quotes are escaped, each field is enclosed between a pair of double quotes, and everything else that is listed in items 1 thru 7 at [](https://tools.ietf.org/html/rfc4180#section-2) is taken care of. Thoughts?

nickrbb, 2021-04-29:
As requested by @gunnarx, I have submitted some proposed changes to vspec2csv.py to address this issue. Please review and let me know of any issues.

erikbosch, 2021-05-03:
@nickrbb - where is the changes, I could not find any PR at https://github.com/GENIVI/vss-tools/pulls?q=is%3Aopen+is%3Apr

gunnarx, 2021-05-03:
Great!  Agree to limit the scope of this ticket to fixing immediate bugs.  We discussed on the last call and concluded that going full unicode requires a bigger project (but I would be in favor of it as a long term thing) -> New ticket #280

> validate the contents of each field based on a set of allowable chars would definitely be useful 

Yes, agreed.

> (e.g. prevent unit8[] appearing as a data type)

This I thought we did already for things like data types that are constrained to a specific list of possibilities?  Maybe a bug here for arrays?

nickrbb, 2021-05-04:
> @nickrbb - where is the changes, I could not find any PR at https://github.com/GENIVI/vss-tools/pulls?q=is%3Aopen+is%3Apr

Oh, well I did a "git commit" on the file "vspec2csv.py" within the "vss-tools" directory, so assumed the updated file would go into some holding area for review. If this is not the case, let me know what procedure I need to follow. Apologies, using git is not my full-time day job.

nickrbb, 2021-05-04:
> Great! Agree to limit the scope of this ticket to fixing immediate bugs. We discussed on the last call and concluded that going full unicode requires a bigger project (but I would be in favor of it [...]

From what I can see using the CSV file and providing filters on the "DataType" field, there are two signals with incorrect data types:
1. Vehicle.Cabin.SeatPosCount with uint8[] in Cabin.vspec
2. Vehicle.OBD.DTCList with string[] in OBD.vspec

SebastianSchildt, 2021-05-04:
Unless I misunderstand you, just want to point out, those datatypes are valid: https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/

-> Arrays

erikbosch, 2021-05-04:
Hi @nickrbb 

This is how I do it

- Create a fork of https://github.com/GENIVI/vss-tools (button in upper right corner)
- Then you clone your own fork (https://github.com/erikbosch/vss-tools in my case)
- Then I typically create a branch from master on my local machine, do the changes/tests I need and finally push the new branch to my fork
- Then from the github web UI I create a pull-request from my own fork to the GENIVI repository, see e.g. https://github.com/GENIVI/vss-tools/pull/55 as an example
- Then your PR should show up and will be discussed at the VSS Tuesday meetings

By the way, as of now we specify that we allow array-types (see https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/). As you wrote it is only used in two places, and I do not really know how well it is supported by tooling

nickrbb, 2021-05-04:
> 
> 
> Unless I misunderstand you, just want to point out, those datatypes are valid: https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/
> 
> -> Arrays

Ah OK, that was my mis-reading of that page then. I withdraw my above concern on the data types for the mentioned signals.

nickrbb, 2021-05-04:
Thanks @erikbosch for your kind patience with me and clear guidance on process. I believe I have now created a Pull request that you and others can review.

danielwilms, 2021-05-25:
fixed in https://github.com/GENIVI/vss-tools/pull/56

closed 2021-05-25 by danielwilms