COVESA/vehicle_signal_specification #274: Copyright in README file?

erikbosch, 2021-04-28:
From a comment in https://github.com/w3c/automotive/issues/379 it seems that the status/ambition of VSS can be misunderstood

` It does show a solution of three particular car producers, or I am mistaken?`

I guess this depends on that 3 OEMs are listed at the top of the README, and there is nothing that states that this is a joint activity managed by GENIVI (involving W3C as well) aiming for standardization, and not just a contribution from the 3 companies listed. Some thought that came to my mind:

- Would it make sense to give a general introduction to VSS already in the README, to avoid the missunderstanding?
- Which organizations shall be listed in the README? Only those that has changed the README, anyone that has contributed anything to the repository or none.
- Compare with vss-tools where we have no copyright, would it makse sense to align the READMEs?

gunnarx, 2021-05-11:
It is somewhat unusual to keep (C) statements in README and the interpretation could in theory be that it applies only to the README text itself, which we think is not the intention here.

Group consensus seems to be that it is not needed to have these statements in README, but companies may want to keep the intended information that companies hold copyright somewhere.  We also said, independently of formal copyright, that keeping a section in the README to give credit to companies that have contributed might be useful.

As the overseer of the GENIVI repositories, I'll have to give this some thought and come back.

danielwilms, 2021-05-18:
@danielwilms, check files WRT BMW (C)

SebastianSchildt, 2021-06-17:
Can be closed, fixed in #291

closed 2021-06-17 by danielwilms