COVESA/vehicle_signal_specification #277: doc (minor): Taxonomies diagram contains spelling error

slawr, 2021-04-30:
The diagram in https://genivi.github.io/vehicle_signal_specification/introduction/taxonomy/ contains a spelling error. Specification is misspelt in the box labelled "Tools and Serialization".

gunnarx, 2021-05-18:
To make it easier for anyone to send a patch...
Did you use some particular programs to generate those images @danielwilms ?   (I am thinking of all of the pictures)
If so, can we have the "source" files? :)

Also I noticed that the [pics directory](https://github.com/GENIVI/vehicle_signal_specification/tree/master/pics) and the [docs-gen/static/images](https://github.com/GENIVI/vehicle_signal_specification/tree/master/docs-gen/static/images) seem to be unnecessary duplicates?

mi-hol, 2022-10-12:
> https://genivi.github.io/vehicle_signal_specification/introduction/taxonomy/

@slawr the link gives a 404 error, is this issue therefore still relevant (and needs an update) or could it be closed?

erikbosch, 2022-10-13:
The error still exists - location of generated documentation has changed due to the rename from GENIVI to COVESA

New location: https://covesa.github.io/vehicle_signal_specification/introduction/taxonomy/
Image source: https://github.com/COVESA/vehicle_signal_specification/blob/master/docs-gen/static/images/taxonomies.png
Included by: https://github.com/COVESA/vehicle_signal_specification/blob/master/docs-gen/content/introduction/taxonomy.md

I assume the source of this image is lost, so I guess the image needs to be recreated or documentation refactored so that this image is no longer needed

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

slawr, 2024-02-28:
Reopened issue as the original issue still remains in the documentation.

As the comment history states the diagram should likely be recreated to allow it to be maintained.

erikbosch, 2024-02-29:
Yes, it is still there but no one has for almost 3 years been interested in fixing it. When looking at https://covesa.github.io/vehicle_signal_specification/introduction/taxonomy/, do we consider the image as needed, doesn't the text below contain the important information, so a possible solution if we think the text below sufficient would be to just delete the image.

erikbosch, 2024-03-07:
As we now have discussed the issue it does not any longer fulfill the criteria of being older than 2 years and inactive for 1 year, so I suggest keeping it open at least a year longer :-)

josesnchz, 2024-03-07:
Guess you could use that file... Not the same font, since I don't know the one the author used, but I think it looks nice. 
[image]