COVESA/vehicle_signal_specification #280: Unicode support

gunnarx, 2021-05-03:
Placeholder for discussion on allowable characters in the YAML definitions.

- Today: What are allowable characters, for example in definition (right now only letters, numbers and "." encoded in ASCII seems assumed).
- Future: What are future rules taking into account non-english languages, as well as the general consensus in many programming environments to use unicode file formats (UTF8 would be a good small step of course).
- Also, make sure to compare to YAML specs so that the result is still YAML compliant.

crea7or, 2021-09-11:
Since JSON and YAML are UTF-8 by default there are not so many choices.

oliveirarleo, 2021-09-13:
Well, I think more support is good, but this may interfere with some things already defined, for example `.` and `#` are already reserve.

Another thing is that someone will be able to use emojis and other characters that may not be desirable.

closed 2024-02-20 by erikbosch