COVESA/vehicle_signal_specification #283: Default only specified vor attributes but also used for actuators/sensors

SebastianSchildt, 2021-05-26:
See title

We define the `default` attribute for attributes,

https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/attributes/

but not for sensors/actuators

https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/sensor_actuator/

However we *do* use them for sensors/actuators:

https://github.com/GENIVI/vehicle_signal_specification/blob/9c82e7018b05958706107bf86a7e30feca970c2b/spec/Powertrain/Battery.vspec#L111

Should we consider this a bug in the documentation or in the spec?

SebastianSchildt, 2021-05-26:
@erikbosch I somehow have the vague feeling, you brought this up in the past already? But can not find it currently

danielwilms, 2021-05-27:
Hi,

those are the non-attributes, which use default:

```bash
--
./spec//Vehicle/Vehicle.vspec-299-
./spec//Vehicle/Vehicle.vspec-300-- CurrentOverallWeight:
./spec//Vehicle/Vehicle.vspec-301-  datatype: uint16
./spec//Vehicle/Vehicle.vspec-302-  type: sensor
./spec//Vehicle/Vehicle.vspec:303:  default: 0
--
./spec//Powertrain/Battery.vspec-107-
./spec//Powertrain/Battery.vspec-108-- Charging.ChargePortFlap:
./spec//Powertrain/Battery.vspec-109-  datatype: string
./spec//Powertrain/Battery.vspec-110-  type: actuator
./spec//Powertrain/Battery.vspec:111:  default: closed
--
./spec//Powertrain/Battery.vspec-127-- Charging.Mode:
./spec//Powertrain/Battery.vspec-128-  datatype: string
./spec//Powertrain/Battery.vspec-129-  type: actuator
./spec//Powertrain/Battery.vspec:130:  default: manual
./spec//Powertrain/Battery.vspec-138-
--
./spec//Powertrain/Battery.vspec-139-- Charging.StartStopCharging:
./spec//Powertrain/Battery.vspec-140-  datatype: string
./spec//Powertrain/Battery.vspec-141-  type: actuator
./spec//Powertrain/Battery.vspec:142:  default: stop
--
./spec//Powertrain/Battery.vspec-180-
./spec//Powertrain/Battery.vspec-181-- Charging.Timer.Mode:
./spec//Powertrain/Battery.vspec-182-  datatype: string
./spec//Powertrain/Battery.vspec-183-  type: actuator
./spec//Powertrain/Battery.vspec:184:  default: starttime
--
```

From the discussions we had earlier, I think we should consider it as bug in the spec. Or any other opinions.

erikbosch, 2021-05-27:
I see no real benefit of having default values for sensors. If the vehicle has not (yet) reported a value to e.g. a VSS-server, then I see it as better that the VSS-server will return an error code telling that the signal is not available (yet) if someone tries to retrieve the value. For actuators I do not even know how the default value shall be handled. Shall the VSS-server e.g. send an automatic request to the vehicle and request that e.g. the ChargePortFlap shall be closed? 

I do not know if I brought it up before but my suggestion is to remove the default-statements from actuators/sensors listed above. 

(Then it is a different discussion if we shall use "default" at all in the standard VSS catalog, even for attributes. But I believe that has been discussed before :-).  )

SebastianSchildt, 2021-05-27:
**/me dons VSS hat**
I have always believed `default` is not useful at all 🔫  . IMHO the intended use case belongs in a (deployment) layer overriding the `value`  .  As that wasn't the consensus so far, in this specific case I would also opt to consider "defaults" for sensors/actuators bugs

**/me dons KUKSA hat**
I don't care, I will implement what the VSS guys think is right, I would however expect doc and spec to be non-contradicting 😸

erikbosch, 2021-06-18:
Has been fixed by #284 right? Ready to be closed?

SebastianSchildt, 2021-06-18:
Yes, fixed. Close.

closed 2021-06-18 by SebastianSchildt