COVESA/vehicle_signal_specification #294: Should there be any empty branches in VSS standard catalog?

gunnarx, 2021-06-08:
While creating the GraphQL schema I found that several branches are defined but they have no leaves attached.

Examples:
```
WARNING: Skipping branch with no children: Vehicle.Media
WARNING: Skipping branch with no children: Vehicle.Powertrain.FuelCell
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank1.Sensor1
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank1.Sensor2
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank1.Sensor3
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank1.Sensor4
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank2.Sensor1
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank2.Sensor2
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank2.Sensor3
WARNING: Skipping branch with no children: Vehicle.OBD.O2Alt.Bank2.Sensor4
```
Noteworthy is that some of the parents have only this empty branch.  So if for example Sensor1 is removed, then suddenly Bank1 has no children.

Question: Let's define a policy for if empty branches are worth defining (for future use or private use?)

danielwilms, 2021-06-22:
remove media & fuelcell, check what OBD branches mean.

gunnarx, 2021-06-29:
Discussion continues in #304.  Need a PR that removes media and fuelcell (until they are ready to be populated with actual info)

PR #365 (2021-11-16): Remove empty branches
Media and Fuelcell branches were empty.
Removed them for now. As they are empty,
no deprecation is needed.

Fixes: #294
commit 6ad36331 (2021-11-16): Remove empty branches

Media and Fuelcell branches were empty.
Removed them for now. As they are empty,
no deprecation is needed.

Fixes: #294
closed 2021-11-16 by danielwilms