COVESA/vehicle_signal_specification #295: How often are attributes allowed to change?

erikbosch, 2021-06-08:
A vehicle might have some attributes that may change during lifetime. For example, the owner might change tire size, or gross weight might be changed for tax or driving license reasons. This is typically changes that happens at most a few times a year.

Does this fit well with our current view on attributes, that they e.g. can be specified when the vspec file is installed (see e.g. https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/attributes/)? Or should they rather be sensors if they might change during the lifetime of the vehicle.

erikbosch, 2021-06-17:
I am not sure if the difference on the "mechanism" for retrieval and update needs to be that different for sensors and attributes. I guess it very much depend on your implementation and your context.  I do not think that we can assume that attribute values always will be read from system memory. Could it not as well be e.g. a Diagnosis System that pushes information to the VSS-server on every start-up?

As an example, for tires it could be possible that size information in the future will be dynamically loaded by reading a RFID (see e.g. https://www.automotive-iq.com/chassis-systems/articles/radio-frequency-identification-rfid-is-coming-to-a-tire-near-you). Then a possible implementation could be a "TireManager" that reads information from the tires and publish/push it to the VSS server. The VSS-server might opt to persist the value, so it can be reused in subsequent power cycles. If the VSS-server does not get any information from the "TireManager" it use the default values specified in the vspec file.

I am thinking if we should change the documentation slightly to make it more visible that attribute values might change, maybe something like below.

https://github.com/GENIVI/vehicle_signal_specification/blob/master/docs-gen/content/rule_set/data_entry/attributes.md

Current:

`The value set for an attribute by a vspec file can be read by a consumer without the need of having the value sent out by a publisher. The attribute, in effect, is configuration data that can be specified when the vspec file is installed.`

Potential update:

`The value set for an attribute by a vspec file can be read by a consumer without the need of having the value sent out by a publisher. Attribute values can also be provided by publishers, similar to sensors. The latter can be useful for attribute values that are likely to change during the lifetime of the vehicle.`

UlfBj, 2021-06-17:
@erikbosch , I find your argumentation and documentation proposal reasonable. 
Maybe one should try to further "quantify" the difference between sensors and attributes in the Potential update?
Something like:
```
The value set for an attribute by a vspec file can be read by a consumer without the need of having the value sent out by a publisher. Attribute values can also be provided by publishers, similar to sensors. The latter can be useful for attribute values that are likely to change during the lifetime of the vehicle. However, this change should typically not be more than once per ignition cycle, or else it should be defined as a sensor instead.
```

PR #301 (2021-07-20): Improving attribute documentation
Also making sure example is aligned with current vpsec definition
and remove double blanks in vspec for related signals.

Fixes #295
commit c09d085f (2021-07-20): Improving attribute documentation

Also making sure example is aligned with current vpsec definition
and remove double blanks in vspec for related signals.

Fixes #295
closed 2021-07-20 by danielwilms