COVESA/vehicle_signal_specification #300: Releases & Release Download Confusing

SebastianSchildt, 2021-06-17:
We should change the way we link releases.

Current situation:

 -  We have a releases page https://genivi.github.io/vehicle_signal_specification/releases/
 That is what most people want (download a .json. csv. etc.)
 - We have "Github Releases" https://github.com/GENIVI/vehicle_signal_specification/releases/tag/v2.1
   These only contain the "source", while we shold have "snapshots" of the source spec for ech release, this is not waht most people need working with VSS

Our Readme links to the latest Github Release

I suggest moving forward changing the Github Releases to either
 - besides the source also contain all the generated files (you can upload arbitrary files, when doing a Github release)
 - contain no files at alle and besides a "changelog" just just pointing to  https://genivi.github.io/vehicle_signal_specification/releases/

Similarly it night be discussed whether the link in the Readme should point to the Github Releases or the Hugo pages.

erikbosch, 2021-06-18:
While discussing the releases page we might also touch upon the news page. It was not updated for 2.1. In the end I believe all of this is related to #292, that we need to agree what to do for each release and document it somewhere

https://genivi.github.io/vehicle_signal_specification/news/
https://github.com/GENIVI/vehicle_signal_specification/blob/master/docs-gen/content/news.md

danielwilms, 2021-06-22:
use github-releases for release and remove hugo list.

commit d5b0c217 (2021-10-19): Move to github releases

Documentation updates in order to move officially to github releases, including:
* removed releases page
* removed news page
* removed quickstart
* removed release content

fixes: #300