COVESA/vehicle_signal_specification #304: OBD - Oxygen sensors and banks - refactoring needed?

erikbosch, 2021-06-23:
**Summary**

In the last meeting we discussed the empty O2Alt branches in OBD. I am by no way an OBD expert but with my understanding our current representation is wrong. Below is a quite lengthy analysis. Is someone actively using the VSS OBD signals as of today? Any feedback from them on the Oxygen signals would be much appreciated.

Is this is an area where we want to do some refactoring now, or do we want to wait until we have a "real user" of the signals?

**Analysis**

Based on https://en.wikipedia.org/wiki/OBD-II_PIDs my understanding is that PID 0x13 and 0x1D is 1 byte bitmasks to represent how oxygen sensors are arranged in a vehicle. Most vehicles have at most 2 banks and use 0x13 to represent which banks/sensors that are present. Some supercars might have more, e.g. Audi S8 with a V10 engine, and then 0x1D is used to represent which banks/sensors that are present. Regardless, OBD supports at most 8 sensors, values can be received in 0x14-0x1B, 0x24-0x2B and 0x34-0x3B. 

References:
https://mechanicbase.com/troubleshooting/bank1-vs-bank2/
https://www.obd-2.de/sensoren.html
https://jhmotorsports.com/o2-or-oxygen-sensor-upstream-bank-4-oem-bosch-for-s6-late-08-and-s8-v10.html
https://github.com/oesmith/obdgpslogger/blob/master/src/obdinfo/obdservicecommands.h

Based on this I think it is wrong to have branches like `Vehicle.OBD.O2.Bank1.Sensor1` and `Vehicle.OBD.O2Alt.Bank1.Sensor1`. If we want to use branches for O2 sensors we should likely instead use something like `Vehicle.OBD.O2WR.Sensor1`, i.e. without specifying bank as that might differ depending on how 0x13 and 0x1D are used.

But then we might have another problem - in OBD the O2 PIDs are 2/4 bytes long, and contain two values

```
0x14-0x1B A: Voltage B: Short term fuel trim 
0x24-0x2B AB: Air-Fuel Equivalence Ratio (lambda,λ) CD: Voltage 
0x34-0x3B AB: Air-Fuel Equivalence Ratio (lambda,λ) CD: Current 
```
... but in our current VSS we just represent the voltage/current. If we want to represent both values then we need to split the values. That could then give us something like:

```
Vehicle.OBD.O2.SensorsPresent2Banks (uint8, 0x13)
Vehicle.OBD.O2.Sensor1.Voltage (float, Volts, 0x14 byte A)
Vehicle.OBD.O2.Sensor1.ShortTermFuelTrim (float, Percent, 0x14 byte B)
Vehicle.OBD.O2.Sensor2.Voltage (float, Volts, 0x15 byte A)
Vehicle.OBD.O2.Sensor2.ShortTermFuelTrim (float, Percent, 0x15 byte B)
...
Vehicle.OBD.O2.SensorsPresent4Banks (uint8, 0x1D)
```

... and then the `Vehicle.OBD.O2Alt` branch could be removed. If you want to know which bank sensor 3 belongs to you must check 0x13 and 0x1D. Note however that we also better should review the  `Vehicle.OBD.O2WR` branch. as the lambda value is not currently represented. That could easily be added.  The lambda is however represented in both 0x24 and 0x34. I do not know if vehicles typically report both current and voltage, or if they typically just report either voltage or current.

```
Vehicle.OBD.O2WR.Sensor1.Lambda (float, 0x24 AB, same as 0x34 AB?)
Vehicle.OBD.O2WR.Sensor1.Voltage (float, 0x24 CD)
Vehicle.OBD.O2WR.Sensor1.Current (float, 0x34 CS)
...
```

SebastianSchildt, 2021-06-29:
I think this needs some careful thinking. However as we already decided the OBD  is about preserving the "semantics"/information of OBD, but not exposing the user to arcane details such as the scaling etc. I do think

 - we should represent values/PIDs that contain voltage/current related to the the same physical entity as branches with two sensors underneath
 - we can/should hide the "bank" bitmasking thing, as it just describes how (in OBD) further values are encoded/to be found, and just list all sensors in VSS

erikbosch, 2021-07-20:
Example of how we could do the change in https://github.com/erikbosch/vehicle_signal_specification/pull/4/files

erikbosch, 2021-08-10:
Discussed briefly at meeting 2021-08-10. Erik to create a "real PR" that can be discussed

PR #324 (2021-09-14): Refactoring OBD oxygen sensors
Fixes #304

This is to my understanding a better representation compared to current one.
I am by no means an OBD-expert so input from anyone with good knowledge on OBD or experience from using the current VSS OBD specification would be great.
commit ec445931 (2021-09-14): Refactoring OBD oxygen sensors

Fixes #304

This is to my understanding a better representation compared to current one
closed 2021-09-14 by danielwilms