COVESA/vehicle_signal_specification #308: [WIP] online doc: broken paths

slawr, 2021-06-25:
I found various broken paths in the online documentation. In the process of completing a PR to fix them.

slawr, 2021-07-22:
@gunnarx @danielwilms shall I just close this now #309 is merged or do maintainers prefer to label and do it themselves? Personally I'm used to just discussing on a PR but wasn't sure what was the preferred norm in VSS so raised issues as well in case it was preferred to have general discussion in the issue and technical discussion in the PR.

danielwilms, 2021-07-25:
@slawr, no worries. I'll close it. I usually link the issue in the PR description, so that it closes automatically.

as general rule of thumb:
* if a change is clear, not much discussion is expected and you can do it yourself, go ahead and do a PR.
* if not open an issue.

closed 2021-07-25 by danielwilms