COVESA/vehicle_signal_specification #310: improvement doc: minimum Hugo version is required but undocumented

slawr, 2021-06-25:
Hi,

Whilst trying out my documentation changes I notice a gap in the documentation build doc. No minimum Hugo version is documented but appears to be required.

Ubuntu 18.04 LTS packages Hugo version 0.40.1. The Learn theme documents the minimum Hugo version as v0.25 (https://themes.gohugo.io/hugo-theme-learn/). However trying to render the VSS doc produces errors such as:
```
hugo server -s ./docs-gen
Building sites … ERROR 2021/06/25 16:00:19 Failed to add template "theme/partials/header.html" in path "/home/steve/genivi/vehicle_signal_specification/docs-gen/themes/learn/layouts/partials/header.html": template: theme/partials/header.html:6: function "hugo" not defined
ERROR 2021/06/25 16:00:19 Failed to add template "theme/partials/menu.html" in path "/home/steve/genivi/vehicle_signal_specification/docs-gen/themes/learn/layouts/partials/menu.html": template: theme/partials/menu.html:107: unexpected "=" in operand
ERROR 2021/06/25 16:00:19 theme/partials/header.html : template: theme/partials/header.html:6: function "hugo" not defined
ERROR 2021/06/25 16:00:19 theme/partials/menu.html : template: theme/partials/menu.html:107: unexpected "=" in operand
```

Reverse engineering the VSS CI I can see it uses for the doc check Hugo version v0.78.1. Installing the same version resolves the errors above so there appears to be some minimum Hugo version required but I can not find anything documenting it in the readme files or online doc. Not being very familiar with the interaction of the markdown used in the doc and the Hugo and Learn versions I can't state what it is so its hard for me to send a PR to correct it. It would be an improvement to the document to do that however to avoid setup errors with newcomers.

Regards

Steve

gunnarx, 2021-06-29:
Discussed today.  We were considering that "user" documentation might not really need to explain how we build our web documentation.  But a README in the docs_gen directory instead?

slawr, 2021-07-02:
Move https://genivi.github.io/vehicle_signal_specification/tools/vss_documentation/ to the README you suggest you mean? Sure you could do that if you wish to separate user and developer documentation. My only comment there would be something to aid discoverability of that new README with the goal of encouraging improvements.

That doesn't address the gap in the developer doc about versions though :). I was thinking later that perhaps there is no need to track some definitive minimum version. A simple v0.78,1 is known to work is better than none.

PR #311 (2021-07-13): Rework "Tools" section in Hugo pages 

- Removed outdated tools documentation from vss doc 
- Moved HUGO docs into Readme.md in docs-gen folder
- Renamed "Tools" in webpage to "Ecosystem" and also added VISS and software links

This will fix #310  and help with https://github.com/GENIVI/vss-tools/issues/74 (removing one source of contradiction) and provides an alternative fix to https://github.com/GENIVI/vss-tools/pull/73 (now all tools documentation is vss-tools, and the vss HUGO page only refers to the vss-tools repo)
closed 2021-07-13 by danielwilms