COVESA/vehicle_signal_specification #321: Improving documentation on how to participate in VSS calls

erikbosch, 2021-08-11:
In the README file we have a short section on "Contribute to VSS", but we do not mention the weekly VSS-meetings. The VSS-meetings are briefly mentioned at https://at.projects.genivi.org/wiki/display/DIRO/Common+Vehicle+Interface+Initiative+--+Home but without more detailed information concerning for example if a specific meeting is cancelled or not.

Some ideas:

- Would it be an idea to have a specific GENIVI wiki-page for the VSS/VSSo meeting?
- That page could in addition to call-in details also give a rough agenda for the meeting, and describe the general process for contributing to VSS, for example encouraging anyone that files an issue/PR to participate in the call.
- The page could also be used to communicate that certain instances of the meeting are cancelled.
- The README file in VSS (and vss-tools) could possibly refer to that page, so that it is easy to find the meeting information.

SebastianSchildt, 2021-08-24:
I like the idea. Maybe @gunnarx can create a (small) Wiki Page with he coordinates and then it can be linked in README.

But it should be a very VSS-(meeting) specific page, not any of those large  CVII page  where I don't think any uninitiated ever found anything :)  Of course it could provide on or two further links for the adventurous

SebastianSchildt, 2021-09-14:
wrt to the PR above: Was not sure, whether to paste the same paragraph to VSS-tools, or whether we just link back to this repo there (maybe in a more prominent place, just a sentence in the beginning, that those are tools for VSS; currently it is "hidden" in paragraph 2)

danielwilms, 2021-09-14:
No other work to be done here.

PR #337 (2021-09-14): README.md: Add link to regular call information
Adding link to regular call information.

Will fix #321
closed 2021-09-14 by danielwilms