COVESA/vehicle_signal_specification #327: Is VSS can be responsible for processing of coupled pairs actuator<>signal?

crea7or, 2021-08-13:
Our team makes some high level objects over the our current API (not VSS based) that sets some state to can device(actuator) and wait for specified amount of time to see if the state has been changed/applied(signal). So, is this functionality can be implemented in VSS(theoretically) or it should be done in the some other level outside VSS?
Possible realization include some timeout and required state, if state is not reached in the specified time - an event/subscription on actuator(or maybe signal) should be triggered to notify client that the signal was not applied.

erikbosch, 2021-08-16:
I think this is in the borderline between VSS and e.g. VISS

We have in VSS today some actuators that cannot be instantly mechanically changed, like the two tilt signals below.

```
Vehicle.Body.Mirrors.Left.Tilt
Vehicle.Chassis.SteeringWheel.Tilt
```

My assumption so far has been that as they are actuators, you can both request a particular tilt, and read the (current or wanted?) tilt using the same signal. I do not if we have previously discussed if the read value of an actuator means the "current value" or the "wanted value".

I assume (correct me if I am wrong @UlfBj ) that in VISS a successful reply on an "update" indicates just that the server accepted the request (possibly including e.g. range check of data), it does not mean that it actually was possible to execute the request. I.e. for the tilt-cases you will get an almost instant response from VISS, but it will possibly take some seconds until the mirror or steering wheel reaches the wanted position. And if there are something that blocks the movement it may not even reach the wanted position, but VISS will not tell you unless you read/subscribe the signal, right?

If we assume that reading always returns the current position one could think of a flow like below, and then it is up to the VISS client to monitor how the movement proceeds. 

1. Read "Tilt" -> Response 0%
2. Subscribe to "Tilt"
3. Update "Tilt", request 100% -> Response "ok". Mirror starts moving
4. Subscription notification: 60%
5. Subscription notification: 70%
6. Read "Tilt" -> Response 72%
7. Subscription notification: 80%
8. Subscription notification: 90%
9. Subscription notification: 100%. Mirror stops moving

The alternative would be that reading the actuator always returns the "wanted" value. But then the actuator might want to reset itself if it fails to perform the action. (example: If an "open_door" actuator fails to open the door because it is frozen, it might give up after a few seconds and reset it self to "false"). For the tilt example - if 100% tilt was requested but the vehicle by some reason only can reach 66% then the actuator value will after a vehicle specific time reset from 100% to 66%

UlfBj, 2021-08-16:
> I assume (correct me if I am wrong @UlfBj ) that in VISS a successful reply on an "update" indicates just that the server accepted the request (possibly including e.g. range check of data), it does [...]

Your description above is how I believe VISS should behave. It may not be clear from the spec, and in that case a description should probably be added. 

>> And if there are something that blocks the movement it may not even reach the wanted position, but VISS will not tell you unless you read/subscribe the signal, right? 

Right. Also this should be clarifies in the spec.

UlfBj, 2021-08-17:
@crea7or  : What you describe I think should be handled outside VSS. From your description it seems like your API can be used to handle it.

crea7or, 2021-08-17:
> @crea7or : What you describe I think should be handled outside VSS. From your description it seems like your API can be used to handle it.

Thank you for your comments and ideas!

closed 2021-08-17 by crea7or

gunnarx, 2021-08-17:
First, it's worth noting that both VSS and the related [VSC/IDL](https://github.com/GENIVI/vehicle_service_catalog) definition work  frequently discusses this issue. (Recommended [introduction slide deck](https://at.projects.genivi.org/wiki/download/attachments/74973446/VCS-IDL%20presentation%20July%201%20ver%208.pdf?version=1&modificationDate=1625161469000&api=v2), also [vsc-tools](https://github.com/genivi/vsc-tools/)).  Some propose that actuation should _only_ be done through a real RPC function call instead, which will "return" something to clarify if the setting the value was successful (even if actuation takes some time).  Others feel that writable (actuator) signals are OK, and can cause a behavior.  For example, using the behavior described above (and in the next comment).  I really recommend getting involved in the VSC/IDL work to work out these issues, and consider combining VSS (for data) with VSC (full function interface / remote-procedure call) as a "solution" to the full system design challenge.

gunnarx, 2021-08-17:
VISS Conclusion:  

Further discussions on today's W3C meeting for defining the VISSv2 protocol decided that the behavior for "actuator" should be that when you write, it is the wanted/target value and the protocol will respond only that this target value was received OK. And when you read (explicitly or through subscription), then it is returning the _current value_, which could still differ from the target value, if actuation takes some time.

BUT PLEASE NOTE!  While this is a reasonable behavior (maybe especially for a stateless HTTP protocol), that decision only defines the behavior of the **VISS v2 protocol** ([core spec](https://raw.githack.com/w3c/automotive/gh-pages/spec/VISSv2_Core.html), [transport spec](https://raw.githack.com/w3c/automotive/gh-pages/spec/VISSv2_Transport.html)).  It is not (currently?) automatically guaranteed to be the behavior for other usage of VSS but this behavior could be a useful guidance for similar situations.    Again, recommend _[getting involved in](https://at.projects.genivi.org/wiki/display/DIRO/Common+Vehicle+Interface+Initiative+--+Home)_ weekly VSS/VSC calls to consider how to work out these questions.