COVESA/vehicle_signal_specification #328: Update of Gear signal declaration

UlfBj, 2021-08-17:
The Vehicle.Powertrain.Transmission.Gear signal declaration does not describe what the values should be for Park and Drive in the case of an automatic gear box. 
Also, some vehicles might have more than one reverse gear, so -2, -3, .. should be supported. 
Current declaration:
Gear:
  datatype: int8
  type: actuator
  min: -1
  max: 16
  description: Current gear. 0=Neutral. -1=Reverse

Proposed updated declaration:
Current declaration:
Gear:
  datatype: int8
  type: actuator
  description: Current gear. 0=Neutral, -1/-2/..=Reverse, 126=Park, 127=Drive,

SebastianSchildt, 2021-08-17:
Hah, recently I came around the same problem.

I do feel, instead of having a description of "magic numbers", we should use strings directly? For an example check this line in a DBC file  https://github.com/joshwardell/model3dbc/blob/24ef335b2518452857232baea29105656fb574ae/Model3CAN.dbc#L4159

which already maps the ints (in a very model specific CAN message), to some strings.  It would feel a bit backwards mapping those back to magic ints in VSS, which aims be a much "higher layer" standard.

erikbosch, 2021-08-18:
One can also discuss if an enum would make sense for this signal (Like in the obsolete example in https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/enum_sensor_actuator/). The signal is an actuator, so how shall the (VISS) client otherwise know which values that are supported by the vehicle, and how shall the VISS-server be able to reject a request to change to gear "-2".

And when I say enum, I mean a definition, possibly on the infamous layer level, which specifies allowed values for this particular vehicle. I do not care if it use int as base like `[ -1, 0, 1, 2, 3, 126, 127]` or programming language friendly enum identifiers like `[ R1, N, G1, G2, G3, P,D]` or arbitrary strings like ` [ -1, N, 1, 2, 3, P,D]`. And for a VISS client the order might actually be important, if you e.g. want to create an application with "+" and "-" buttons which have the same meaning as the corresponding buttons in the car.

UlfBj, 2021-08-18:
The integer values in my proposal are not picked by me, but comes from our experience "in the field". 
The proposal by @erikbosch is aligned with my proposal, I think something along those lines would be ok.

oliveirarleo, 2021-08-18:
I believe that this case can be correlated to what I said in the other issue:
https://github.com/GENIVI/vehicle_signal_specification/issues/323#issuecomment-901113823

That solution proposed would solve the correlation said in the description: `0=Neutral, -1/-2/..=Reverse, 126=Park, 127=Drive,`

erikbosch, 2021-08-18:
Concerning Park/Drive values, the values 126/127 seems to be aligned with e.g. SPN525 in J1939 if one consider the offset used there.

A side comment - we should for this signal better document that the signal is for "requested gear" and not "actual gear". I.e. if you have the gear stick in "Drive" mode then it shall return "Drive", not the actual physical gear (e.g. 4) currently engaged by the transmission

danielwilms, 2021-08-31:
Do we have to distinguish between gear number and mode (mode is definitely not the best term). Even if you're driving automatic in `D` mode it might be interesting in which number you're in?

UlfBj, 2021-08-31:
In Android automotive there are two signals, selected gear, and actual gear.

danielwilms, 2021-08-31:
@UlfBj creates PR with 2 signals

danielwilms, 2021-09-14:
PR merged #335

closed 2021-09-14 by danielwilms