COVESA/vehicle_signal_specification #329: Default value behavior for arrays

crea7or, 2021-08-20:
If datatype is specified as an array of some types, why default value is not an array, even if it's a single element?  Maybe this is an typo, but documentation isn't clear in this case too. 

example from generated json:

```json
"SeatPosCount": {
            "datatype": "uint8[]",
            "default": 0,
            "description": "Number of seats across each row from the front to the rear",
            "type": "attribute",
            "uuid": "8dd40ecd47ab51c79ed9c74ae4296d7e"
          }
```

erikbosch, 2021-08-23:
I think this is a typo/regression introduced in https://github.com/GENIVI/vehicle_signal_specification/commit/2714a7a0d156aa12ffd38bdf5c418241e1f1ec7a 

One can argue that the default value shall be removed from the VSS-spec as that must be specified for each vehicle on "layer" level. But still - an interesting question how a default value for an array shall be expressed on "layer" level. Should it e.g. be defined as `[2, 3]` for a standard car with 2 seats in the front row and 3 in the back row?

crea7or, 2021-08-23:
Currently we are trying to transfer some non-can related configs to vss (because it's suitable for it) and default values is a must have. Maybe a tag name is not a very good since attributes should have just read-only "value", but vss allow to change them so it's ok.

danielwilms, 2021-08-31:
@erikbosch, make proposal for documentation.

PR #334 (2021-09-07): Changing seat instantiation and improving default value documentation
Content of this change:

- Improved documentation on how to set default values for array
- Changing default values for seats/doors. Now default corresponds to a classical passenger car with 2 rows of seats, 2 seats in front row and 3 in back row. That reduces the size of generated "standard VSS" significantly.

Fixes #329

Example on how generated JSON with new default looks like:
```
 "SeatPosCount": {
            "datatype": "uint8[]",
            "default": [
              2,
              3
            ],
            "de [... cut]
commit 247fdeeb (2021-09-07): Changing seat instantiation

Also improving documentation of default
and changing comments on some signals

Fixes #329
closed 2021-09-07 by danielwilms