COVESA/vehicle_signal_specification #330: Definition in vspec of Min/Max/Default values for float and double

crea7or, 2021-08-22:
The float values in the current json output is defined visually as int. It's absolutely legal in json and yaml, since they does not differentiate numbers by type neither the python do, but some c++ json libraries can check it and we can use it as additional validation step. Maybe it will be better to define float values as explicitly float like 0.0 or 90.0? It can be done in the current vss generators ( vspec2*.py ).

current:
```json
"AverageConsumption": {
                "datatype": "float",
                "description": "Average consumption in liters per 100 km.",
                "min": 0,
                "type": "sensor",
                "unit": "l/100km",
                "uuid": "e2252108125a54dcab34e1bad0fe8bdc"
              },
```
explicit float:
```json
"AverageConsumption": {
                "datatype": "float",
                "description": "Average consumption in liters per 100 km.",
                "min": 0.0,
                "type": "sensor",
                "unit": "l/100km",
                "uuid": "e2252108125a54dcab34e1bad0fe8bdc"
              },
```

oliveirarleo, 2021-08-23:
That makes sense for me, since `min`, `max`, and `default` values are contained in the set of `datatype`. Is the `AverageConsumption` leaf the only case that this happens?

crea7or, 2021-08-23:
nope, full list after my validator:

Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Heading -> min/max value is not double: int64
Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Latitude -> min/max value is not double: int64
Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Longitude -> min/max value is not double: int64
Vehicle.Cabin.Infotainment.Navigation.DestinationSet.Latitude -> min/max value is not double: int64
Vehicle.Cabin.Infotainment.Navigation.DestinationSet.Longitude -> min/max value is not double: int64
Vehicle.CurrentLocation.Heading -> min/max value is not double: int64
Vehicle.CurrentLocation.Latitude -> min/max value is not double: int64
Vehicle.CurrentLocation.Longitude -> min/max value is not double: int64
Vehicle.Powertrain.Battery.StateOfCharge.Current -> min value is not float: int64
Vehicle.Powertrain.Battery.StateOfCharge.Displayed -> min value is not float: int64
Vehicle.Powertrain.FuelSystem.AverageConsumption -> min value is not float: int64
Vehicle.Powertrain.FuelSystem.InstantConsumption -> min value is not float: int64
Vehicle.cargoVolume -> min value is not float: int64

SebastianSchildt, 2021-08-24:
I am undecided on this one. Basically I consider any "int" a valid float.

Whether it is "good" or "bad" depends on taste and techstack I guess: In explicitly/statically typed languages such as C `float f=0`is fine and unambiguous. Dynamically typed things such as python also can work with that. 

The only pitfall I can see is, if you end up doing arithmetic depending on the semantics of your language you accidentally end with integer arithmetic instead of float.

If fixing/changing this I think tools/generators is the right place, as in the original vspec YAML, as pointed out by the OP it is perfectly valid, and we should not restrict YAML. But we can encourage good style of course :)

So **tldr;**  You discuss, I agree 😀

SebastianSchildt, 2021-08-24:
Discussed 08/24:  We do not consider it a "bug" atm, but we would likely accepting PRs either changing the original vspec files and/or the output of our tooling.

gunnarx, 2021-08-24:
OK, so to clarify we then require tools to correctly handle "0" and similar, as vspec input even if the type is float.   

I think it is fair to say that production-level implementations should normally not blindly copy the text of the input - they should validate that it is a number and not garbage, and therefore handling "0" as a valid float is just another requirement.

Tools that _output_ an intermediate format (for example vspec2json, or the proposed vspec2yaml if it is accepted) could choose to output "0.0" if the type is float.  That sounds reasonable to me.

danielwilms, 2021-08-31:
So, I think we found a conclusion and can close this one?

closed 2021-09-07 by gunnarx