COVESA/vehicle_signal_specification #332: Handling multiple data points in the signal

crea7or, 2021-08-23:
It's not clear how to handle multiple data points in the signal entry (history values). Should the signal datatype be an array for that or this is an orthogonal features?

erikbosch, 2021-08-24:
VISS has support to request historical values for a VSS data signal, but it then requires that the server actually saves historical values. 

https://rawcdn.githack.com/w3c/automotive/3b84ac8b41586c1dbfc83aed5ca956a77a9f1445/spec/VISSv2_Core.html#history-filter-operation
https://rawcdn.githack.com/w3c/automotive/3b84ac8b41586c1dbfc83aed5ca956a77a9f1445/spec/VISSv2_Transport.html#https-history-read

crea7or, 2021-08-24:
So there are no any sign in VSS that signal should keep historical data? It's a bit confusing who should know is there any data to receive or not and how server will know that historical data must be kept.

SebastianSchildt, 2021-08-24:
I would assume, in the end it is an application decision which data _should_ be kept as history, I do not think it can be "decided"/specified in VSS level (it is not only about which data points, but also about how long that history should be etc.)

VISS defines the (maybe optional?) feature of requesting historical data, which is for example supported by the VISS reference implementation  https://github.com/MEAE-GOT/W3C_VehicleSignalInterfaceImpl , but in contrast is not supported (and not intended to be supported) in the kuksa.val VISS server https://github.com/eclipse/kuksa.val

Apart from VISS, VSS might be used in other systems/tech stacks which would not necessarily have the need or capability for historic data.

For VISS you can see here how it would encode multiple datatpoints:
https://rawcdn.githack.com/w3c/automotive/3b84ac8b41586c1dbfc83aed5ca956a77a9f1445/spec/VISSv2_Core.html#multiple-signals-request, i.e. "Request for multiple values from a single node."
```
		“data”:{“path”:”X”, “dp”: [{“value”:”Y1”, “ts”:”Z1”}, …, {“value”:”Yn”, “ts”:”Zn”}]}
```
A list of datapoints `dp` whereas each datapoint is a `value`/ `ts`(timestamp) tuple (explicitly expressed as json object)

danielwilms, 2021-08-31:
> I would assume, in the end it is an application decision which data should be kept as history, I do not think it can be "decided"/specified in VSS level (it is not only about which data points, but also about how long that history should be etc.)

Agree to that. In my opinion, it shouldn't be part of VSS, even though I can see the need on the protocol side.

gunnarx, 2021-08-31:
> So there are no any sign in VSS that signal should keep historical data? 

That's right.  The "VSS project" as it is defined today only deals with defining the data items.  There are many things the VSS itself does NOT define (but related projects might still do):

 - The protocol for how to fetch data (e.g. by VISS or other protocol)
 - How to serialize/encode the data itself.  It only states the datatype (e.g. int) that is expected to be input and output from a data transfer pipeline.
 - If only the latest value should be kept, or a history of values
 - Quality information about the signal ...  and probably a lot of other ideas (see layers below).

Now, with that said, in the larger discussion (including the CVI Initiative, other subprojects, etc.) these types of questions are intended to be solved.  We are all together trying to work out as many of the details as possible to provide the tools for a full system design.

> It's a bit confusing who should know is there any data to receive or not and how server will know that historical data must be kept.

Before I continue, note that one part of the answer is in the VISS specification.  For that specification (which is one choice of protocol for VSS but does not have to be the only one), it includes the ability to fetch historical values *if* they exist.  Please read the spec for more details.

So then, my opinion is just reinforcing what others wrote:  This must be defined at some later time.  It should not be part of the core VSS, because it will surely be different in different systems, and should not require changing the VSS model itself.  

But here is a list of possibilities (I am sure there are more):

**1.** A protocol specifies some way of setting the length of desired history for a particular signal through a protocol request, and then the system will be from that point on keeping that many values.  It is up to the protocol specification to decide this and you are welcome to propose new protocols, or changes to VISS.
**2.** A protocol includes the ability to dynamically request a "measurement job", i.e. something like: _"I want you to now start recording every change in this signal X.Y.Z for the next 2 minutes."_, or "If condition A > B occurs then start recording these two signals", and the protocol would then give some ability to fetch that collected data later on.  (Often known as a "flight recorder" functionality).  I do not know of  such a protocol at this point, but I certainly envision something like it that will use VSS definitions for the signal "X".
**3.** It is in some cases maybe just defined in the system itself (with no external config file, simply programmed that way).  It might be decided by context and system, etc.

or
 
**4.** A particular metadata tag _could be added_ to VSS, but then in separate layer.  (A layer is simply an additional VSS-formatted input file that adds more metadata, or changes existing definitions).   In this scenario it is _kind-of_ added to VSS as you request here, but not in the core model but in a layer.
Then, when it comes to layers, we basically have two options:
   **A.** Don't specify anything in the shared "VSS project" - let any adopter of VSS create their own _proprietary_ metadata label.  You are able to do that in your own development if you wish, define a field that specifies the size of the remembered history.

or

   **B.** Specify a shared (required/recommended or proposed) metadata label and meaning.  This is very similar to the metadata that is part of normal VSS, it has a shared understanding of what it means, but it is proposed to be part of a layer (and is of course optional to include, which most of the "core" VSS metadata is not).

As a collaboration we have not yet come to a point **B** where "recommended" layer content is defined.  But there are some specific converters coming soon that will need additional mapping information between VSS and non-VSS signals, so there will soon be some "defacto" agreements on additional metadata in layers.   If you come up with a proposal on the protocol to be used together with this "history size" item, then perhaps a suggestion can be made to include it (again, in layers, not in core).

gunnarx, 2021-08-31:
For clarity I'll try to also answer some of your specific points @crea7or 
> Should the signal datatype be an array for that or this is an orthogonal features?

Generally, it should not _misuse_ the array feature.  It is an orthogonal concept as you say.
 A signal is of type array only if for some reason it needs to carry multiple things seen as a single concept (signal name).

I think it is better if VSS define the fundamental signal, not multiple instances (measured values) of that signal.  So most VSS signals should end up being a single thing, integer, string etc.   

In certain cases, an array type is useful however.
@erikbosch pointed out today that there might be some exceptions such as pecific data items that are intentionally defined that way - as an array of related values.  (Even a "history" in this case, but it is only a specific decision for a specific signal, a special case).

The made-up example we discussed:
```
X.Y.Z.Diagnostics.Last10TroubleCodes
              datatype:  uint32[10]
              description:  This signal includes the last 10 trouble codes, in timestamp order, with the most recent one at index 0.
```

... but note that this is explicitly decided to be defined as one "signal" (VSS data item) that just happens to contain 10 items.  It is not a generic methodology for deciding a history of values measured for any VSS signal.

Remember that it is tricky to keep apart the definition of a VSS data item / signal (which is what VSS defines) from the _measured values_ of that signal (which could be any number of...).   

> We have already implemented "data"/"data points" feature already. The question is more about signals.

I realize now there is a very big risk we don't understand each other's meaning of "data points" or "signals" here.
Hopefully you get it in my text here, otherwise we should discuss more carefully.

>  I think that array can be a sign to server to not to replace value of the signal/actuator in the internal data structure, but push_back it to the array. It's very easy, clean and controllable. I think that this is the right layer to keep the history since everything else do allow use it in a such way. Without history filter server will return last value, with filter - the entire array. With curve logic array should be fixed and it will trigger notification. Without exact pointing to the history feature, requesting the history from a some random signal does lead us to many-many questions, how to do it and what to return at the first call and where to keep it.

I disagree that this is clean.  We have, hopefully, established here that the concept of the array datatype is not intended to indicate how many historical _measured values_ will be kept for each VSS item.    (I write VSS "item" here, but we often would call it a VSS signal, but I'm afraid you might work with a definition where "signal" means something like a sent "message" or "measured  value"?  which is not the intention.  It is simply the definition of the data item I mean when I talk about a VSS signal.  (Signal = an observable data property)

crea7or, 2021-08-31:
@gunnarx thank you very much for such a good clarification. I think that I get it.

closed 2021-08-31 by crea7or