COVESA/vehicle_signal_specification #342: Incorrect delimiters in `docs-gen/static/releases/*/*.csv` files

cwelton, 2021-09-29:
When including strings that have commas in a csv file the string needs to be quoted, otherwise the .csv file cannot be correctly parsed.

Example:

Bad:
```
Vehicle.VehicleIdentification.bodyType,attribute,string,,false,,,,Indicates the design and body style of the vehicle (e.g. station wagon, hatchback, etc.).,,3f6910e6773c532e986240e3d45f623b,
```

Better:
```
Vehicle.VehicleIdentification.bodyType,attribute,string,,false,,,,"Indicates the design and body style of the vehicle (e.g. station wagon, hatchback, etc.).",,3f6910e6773c532e986240e3d45f623b,
```

erikbosch, 2021-09-30:
Which version of vss-tools are you using? There has been a recently new change (https://github.com/GENIVI/vss-tools/pull/56/files) to quote all fields in generated CSV files.

danielwilms, 2022-02-27:
We can fix that for the older versions, as it should be a rather small fix.

erikbosch, 2022-08-23:
The data is correctly quoted in latest release (VSS 3.0). As no-one seems to volunteer to patch earlier releases I will close this issue.

closed 2022-08-23 by erikbosch