COVESA/vehicle_signal_specification #364: Adding support of "comment"?

erikbosch, 2021-11-15:
In VSS we have today `description`, and we can give unstructured comments by using `#`. I am considering if it would be beneficial to extend our language definition so that also structured comments could be given. Then we can have the recommendation that description shall be rather short, more like a title, additional information shall be put in comments.

If we think this would be a good idea the first step would be to change documentation and generic tooling so that `comment` is accepted and consumed. Then I would consider it optional for tools (csv, json, ocf, protobuf, ...) if they want to include comments in their output or not.

Example of a current signal:

```
DistractionLevel:
  datatype: float
  type: sensor
  unit: percent
  description: Distraction level of the driver will be the level how much the driver is distracted, by multiple factors. E.g. Driving situation, acustical or optical signales inside the cockpit, phone calls
```

Possible usage if comment is used:

```
DistractionLevel:
  datatype: float
  type: sensor
  unit: percent
  description: Distraction level of the driver.
  comment: Can be calculated by multiple factors E.g. Driving situation, acustical or optical signales inside the cockpit, phone calls
```

danielwilms, 2021-11-16:
Thanks @erikbosch for the suggestion. I absolutely think we should find a solution for it, as otherwise the information in the unstructured comments is 'lost'. I like `comment` as keyword. It's used as well in rdfs.

erikbosch, 2021-12-21:
All PRs merged, closing it

closed 2021-12-21 by erikbosch