COVESA/vehicle_signal_specification #368: Wheel.Speed

jdacoello, 2021-11-17:
Can you add to the specification the speed property of a wheel? Look, for example, real data sets like [comma2k19 ](https://github.com/commaai/comma2k19) which have this data for all the wheels {front left, front right, rear left, rear right}

danielwilms, 2021-11-17:
@jdacoello, sure it's possible if there is a use case for it (dataset proves that there is :)). Do you want to create a PR, then we can discuss it there.

erikbosch, 2021-11-17:
Would fit quite well to `Wheel.vspec`, possibly something like below

```
Speed:
  datatype: float
  type: sensor
  unit: km/h
  description: Vehicle speed calculated from wheel rotation.

```
That would give us 4 new signals when expanded:

```
Vehicle.Chassis.Axle.Row1.Wheel.Left.Speed
Vehicle.Chassis.Axle.Row1.Wheel.Right.Speed
Vehicle.Chassis.Axle.Row2.Wheel.Left.Speed
Vehicle.Chassis.Axle.Row2.Wheel.Right.Speed
```

erikbosch, 2021-11-23:
@jdacoello please create a PR and we can discuss it on the next meeting

erikbosch, 2022-08-10:
Fixed by https://github.com/COVESA/vehicle_signal_specification/pull/464

closed 2022-08-10 by erikbosch