COVESA/vehicle_signal_specification #370: Document meaning of writing actuator ("target value")

gunnarx, 2021-11-23:
I think we agreed in a VSS meeting (although the question came up in a VISS meeting?)
that setting an actuator value should generally be considered as setting the "target" value and that this should be the "opinion" of the original VSS specification as guidance to all protocol implementations.

But I didn't find it in the documentation, or is it there?

I think this line  might give another understanding, or at least an incomplete understanding:
From [data_entry/_index.md#L18 
](https://github.com/COVESA/vehicle_signal_specification/blame/bbad4c4538e894f898ecedc83ec608bf14daee19/docs-gen/content/rule_set/data_entry/_index.md#L18) :

> rule_set/data_entry/_index.md:  
>     actuator: actuating signals describe current state of the vehicle and change, when the state of the vehicle changes. **Writing the value, leads to a change of the vehicle state itself (e.g. door lock).**

erikbosch, 2021-11-23:
Meeting: @erikbosch to create PR

erikbosch, 2021-11-24:
Fixed as part of #371

erikbosch, 2021-12-21:
@gunnarx  - this one can be closed now, right?

danielwilms, 2022-02-27:
I think this text from #371 as @erikbosch  fixes the issue:
> Actuators are used to control the desired value of a property. Some properties in a vehicle cannot change instantly. A typical example is position of a seat or a window. Reading a value of an actuator shall return the current actual value, e.g. the current position of the seat, rather than the wanted/desired position. A typical example could be if someone wants to change the position of a seat from 0 to 100. This can be changed by setting the corresponding actuator to 100. If the actuator is read directly after the set request it will still return 0 as it might take some seconds before the seat reaches the wanted position of 100. If the seat by some reason is blocked or cannot be moved due to safety reasons it might never reach the wanted position. It is up to the vehicle to decide how long time it shall try to reach the desired value and what to do if it needs to give up.

Will close it for now. @gunnarx, please reopen, if information is missing

closed 2022-02-27 by danielwilms