COVESA/vehicle_signal_specification #375: Feature request: support strings in the list of suffixes in an instance block

slawr, 2021-11-29:
The VSS v2.2 data model rules support instances using numerical suffixes.
For example:
```
Door:
  type: branch
  instances:
    - Row[1,4]
```
    
producing:
```
Row1.
Row2.
Row3.
Row4.    
```

The request is to extend the rules to allow a list of strings to be used as the suffix.
For example:
```
Door:
  type: branch
  instances:
    - Front["Left", "Right"]
    - Rear["Left", "Right"]
```
would result in:
```
Door.FrontLeft.
Door.FrontRight.
Door.RearLeft.
Door.RearRight.
```

erikbosch, 2021-11-30:
For the sake of completeness, we already support enums like this:

```
Door:
  type: branch
  instances:
    - ["Front","Rear"]
    - ["Left","Right"]
  description: All doors, including windows and switches
```

Will give you signals like:

```
Vehicle.Cabin.Door.Front.Left.Window.ChildLock
Vehicle.Cabin.Door.Front.Right.Window.ChildLock
Vehicle.Cabin.Door.Rear.Left.Window.ChildLock
Vehicle.Cabin.Door.Rear.Right.Window.ChildLock
```

But we do not support prefix in combination with enum, like this:
```
Door:
  type: branch
  instances:
    - ["Front", "Rear"]
    - Pos["Left", "Right"]

```

I assume it should be quite easy to change tooling so that the prefix (if any) is considered and correctly handled.

danielwilms, 2022-02-27:
Personally, I like the notation, we have now, like
```
instances:
    - ["Front","Rear"]
    - ["Left","Right"]
```
more than the prefix from the [first comment](https://github.com/COVESA/vehicle_signal_specification/issues/375#issue-1066319459). But definitely open for discussion. As I see it now, there isn't anything, we don't support with the current solution, right?

erikbosch, 2022-03-30:
Concerning the lack of Airbag for rear seat in your instantiation we have briefly touched the subject of "pruning" the signal tree in e.g. #252  to remove nodes/branches that will not exist in a deployment. A typical example concerns seats where we typically define 2 rows and 3 positions, but many cars does not have 3 positions/seats in the first row, and then tree pruning could be used to remove "Seat.Row1.Pos3".

But as of today we have no proposed way to remove individual signals for individual instances in a deployment, like keeping the airbag branch only for seats in the front row, more than first expanding the tree and then manually removing signals based on what each seat support. But possibly the prune-concept could be extended to cover this as well.

slawr, 2022-03-30:
Thanks for the heads up Erik. To be honest I was thinking later last night that I probably shouldn't have mentioned the airbag in my recent comment as it was in fact a separate issue and that it could do with some editing for simplicity. Dangers of updates at the end of a day. Anyway I will try to revisit it. It's really just "should tooling be able to join two string parts to make a single instance naming".

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch