COVESA/vehicle_signal_specification #377: Use of "default" for attributes/sensors/actuators/enums

erikbosch, 2021-12-02:
As of today `default` is only allowed for attributes. The original VSS intention with default as I understand it was for attributes only, so that the "vehicle data broker" (or whatever we like to call it) would get a value for the attribute already when parsing/importing the VSS, so that no other entity/producer in the vehicle needs to provide that value to the "vehicle data broker" . 

In the VSS Slack channel it has been discussed it it would be relevant to make `default`mandatory for all enum-entities in VSS, i.e. also sensors and actuators. Background is that it is quite common in other contexts that enums must have an explicit/implicit default value.

Topics to discuss:

- Do we want to allow `default`also for sensors and actuators?
- If so only for "enum"-types, or also for integer/float types?
- Do we want to make `default` mandatory for enums?

If we want to make `default` mandatory for all enums (also sensors/actuators) we should in my view first agree on what that means. If we take the actuator below as example. Assume an app wants to read or subscribe to `HMI.DistanceUnit` and the vehicle has just restarted so that the "vehicle Data Broker" is up and running, but the HMI has not yet published any value for `DistanceUnit`. What should then be returned? I assume specifying "km" or "mi" as default could give problems, as the app then first might get "km" (default) and then a few seconds later "mi" when HMI has reported the real value. If we want to make default mandatory for all VSS enums we might consider if the default value for signals/actuators should correspond to undefined/unknown. That would allow for apps to ignore observations reporting the default value.  

Current signal:

```
HMI.DistanceUnit:
  datatype: string
  type: actuator
  enum: [ "mi", "km" ]
  description: Distance unit used in the current HMI
```

Possible refactoring to include default and unknown:

```
HMI.DistanceUnit:
  datatype: string
  type: actuator
  enum: [ "unknown", "mi", "km" ]
  default: "unknown"
  description: Distance unit used in the current HMI
```

SebastianSchildt, 2021-12-14:
Just linking related issues in kuksa.val, so we remember change again if necessary 

https://github.com/eclipse/kuksa.val/issues/201

danielwilms, 2022-02-27:
In the `enum`/`allowed` discussion, we discussed against a mandatory default value. This leaves this question unresolved still:

> Do we want to allow default also for sensors and actuators?

I don't see, why we should limit default to attributes. My general concern is, that we define `default` in the standard catalogue, even though it cannot be generalized and therefore should be part of the concrete deployment of the tree.

IMHO we should define it for all leaves, but use it sparsely in the standard catalogue.

SebastianSchildt, 2022-02-28:
> use it sparsely in the standard catalogue.

Fully agree with that.

Other than that, when used in a "deployment", especially for attributes, I am not even sure "default" is a good name for that, why not "value".?
However, note that I am probably "biassed by implementation", because that is basically what we do in KUKAS.val when encountering `default`, we copy it to `value`

https://github.com/eclipse/kuksa.val/blob/d699105b48c2d11163820fded5bb3f1d2b0a3784/src/VssDatabase.cpp#L76

I am fine either way

danielwilms, 2022-03-29:
**`default` usable for sensor/actuators:**   
* decision 29/03/22 extend it to all leafs
* semantics similar enough to keep one key word? attributes might be different than sensors

**`default` vs `value` vs `initial`:**   
* `value` too broad, 
* default might be misleading as well.
* other suggestions?

slawr, 2022-03-29:
As noted in today's VSS call here are some examples where default could be useful for sensor/actuator values:

1. Traffic intersection: https://github.com/slawr/vss-otaku/blob/0f4c7b8f7f14d64a5f5cbfc1a1e66048a00ee5d9/converters/r-sim2vss/model/renesas-sim-data-model.vspec#L111
2. Transmission gear enum: https://github.com/slawr/vss-otaku/blob/0f4c7b8f7f14d64a5f5cbfc1a1e66048a00ee5d9/converters/r-sim2vss/model/renesas-sim-data-model.vspec#L216
3. Wiper status string enum: https://github.com/slawr/vss-otaku/blob/0f4c7b8f7f14d64a5f5cbfc1a1e66048a00ee5d9/converters/r-sim2vss/model/renesas-sim-data-model.vspec#L336

Note: these all come from a non-VSS data model specified in vspec.

erikbosch, 2023-05-04:
Can be closed as the PR is merged

closed 2024-01-04 by erikbosch