COVESA/vehicle_signal_specification #383: Discuss changing / extending definition of enum

gunnarx, 2021-12-11:
Issue to document the discussion on changing / extending definition of enum.
Don't have the opportunity to summarize the essence of the questions/proposals now, but it could be done by someone...

gunnarx, 2021-12-11:
References:
#381(PR) [SPEC] Changed enum to allowedValues
#377 (Issue) Use of "default" for attributes/sensors/actuators/enums

gunnarx, 2021-12-11:
Notes taken at previous meeting.  Note that these are rough and do not represent all possible proposals or details.
These might be edited later for clarity.

----

What we have today:
```
Charging.ChargePlugType:
  datatype: string
  type: attribute
  default: ccs
  enum: [ 'type 1', 'type 2', 'ccs', 'chademo' ]

Something.Foo:
   datatype: int32
   enum: [ 1, 4, 5 ]
```
Only strings and ints according to documentation.  
Float is not supported?
=> Agreed it should be _possible_ (it seems)
```
   enum: [ 1.5, 2.0, 2.5 ]
```
Consensus(?) Current concept ought to support all types.

---------------
New ideas:

Proposal 1 - similar to enums in C-like languages.
```
Charging.ChargePlugType:
   datatype: int32
   enum [ 'symbol1' : 18, 'symbol2' : -5 ]
```
If no values given:  ` enum [ 'symbol1', 'symbol2']` would assign an integer representation automatically. (0,1,2,3 etc.)

------

Basic agreement that we should note _remove_  current behavior, because it is already used in catalogs / deployed in systems.

Option:    Same behavior as today but renamed to 'Allowed_Values' (or similar).
   - some people like this...
   - some think "enum" is better name, even with this behavior...

----
Opinion:  Limitations for naming are common in programming languages: e.g. start with letter, don't use spaces Do we want those same rules?

----
Opinion: Current enum feature is already used -  remove or change functionality completely would cause problems.
(but renaming seems acceptable)

----
Option "Full featured".  Symbolic names, with optional assignment to value, which includes _all_ types.

```
Something.Foo:
  datatype: string
  type: attribute
  default: ccs
  enum: [ 'type 1', 'type 2', 'ccs', 'chademo' ]
```

Note, if all types are acceptable the consequence is it would possible to have string _values_ assigned to symbolic names.  Where value not equal to the name.

```
Something.Foo:
  datatype: string
  type: attribute
  enum: [ 'symbol1' : 'ComplexStringConstant', 'symbol2' : ' SomeOtherComplexStringConstant' ]
```
(sidenote) --> Note that unlike JSON, names are not normally quoted strings in YAML.  
E.g. `symbol1 : "value"`,  is more YAML like than `"symbol1" : "value"`

----

Also to discuss:  How to, or if we should,  define the Default enum value
Ref: Protobuf proposal (Ulf, on chat)  and/or first value is always default or no default, etc.

----

Opinion:  The full-feature enum should exist, even if it is only recommended to use in Layers.
In standard catalog it is often less useful because real values differ on different systems...

----

danielwilms, 2021-12-14:
Meeting 2021-12-14:
We would need some example on how it could be used and where it's needed.

erikbosch, 2022-09-23:
By the change to "allowed values" I think this one can be closed

closed 2022-09-23 by erikbosch