COVESA/vehicle_signal_specification #392: Add signal for autonomous driving level

gunnarx, 2022-01-04:
I don't have a proposal yet but want to get initial feedback:

**Add autonomous driving level according to SAE defined "levels"** 

Here is one definition, I assume it is valid.
https://orise.orau.gov/resources/stem/documents/scholarships/sae-international-levels-of-autonomous-driving-and-sensor-packages.pdf
(This shows 1-5 but some definitions include zero = "no automation at all")

Rationale:

We don't have a lot of detailed AD signalling in the VSS standard catalog, but this "level" of autonomy is a very basic information many might be interested in seeing in the VSS standard catalog.  It was also _specifically_ requested by OPIN / insurance work.

gunnarx, 2022-01-04:
Added, discussed on today's meeting:

We probably want at least two:
1. What AD level the car is capable of (maximum, attribute).
2. What level the car is currently operating in (real-time, signal).

SebastianSchildt, 2022-01-10:
Agreed. The signal is even more important than the attribute: Depending on the current active level, it might influence what the driver is allowed to do (how distracted can he be? Playing games allowed etc?). In case of a crash, it is be interesting to know which mode the vehicle was operating in

The attribute is also nice (just to complete the "spec sheet" of a car as delivered by VSS), but does not tell you so much. If it claims Level-3 you still would not know "where". Is it only on highways? Up to a specific speed? Ramp up/down included? 
But useful information anyway.

Question is shall this go under Vehicle or ADAS (that branch needs some love anyway)?

SebastianSchildt, 2022-01-11:
Just to get the discussion started

```yaml
Vehicle.ADAS.ActiveAutonomyLevel
  datatype: string
  type: sensor
  default: sae-0
  allowed: [ 'sae-0', 'sae-1', 'sae-2', 'sae-3', 'sae-4', 'sae-5' ]
  description: Indicates the level of autonomy the car is currently operating under. Currently supports levels as defined in the SAE taxonomy
```

for the attribute

```yaml
Vehicle.ADAS.SupportedAutonomyLevel
  datatype: string
  type: attribute
  default: sae-0
  allowed: [ 'sae-0', 'sae-1', 'sae-2', 'sae-3', 'sae-4', 'sae-5' ]
  description: Indicates the level of autonomy the vehicle is supporting. Currently supports levels as defined in the SAE taxonomy
```

Decided against an "int" for the level, because probably other standards/taxonomies will pop up, and it will be easier to extend an int

SebastianSchildt, 2022-01-14:
Made a suggestion as PR in #400 , also following the new naming conventions for `allowed` values

SebastianSchildt, 2022-01-25:
Can be closed for now @gunnarx  @danielwilms

erikbosch, 2022-02-08:
Can be closed for now @gunnarx @danielwilms