COVESA/vehicle_signal_specification #393: Define/Rework Vehicle.EmissionsCO2

SebastianSchildt, 2022-01-11:
Came up in #389 

Wie have an attribute CO2Emissions, but at least the description is lacking

daniyalyasin93, 2022-09-25:
Could you give me a hint where can I find info to contribute in this matter? I have access to some arxml files but they didn't contain any CO2 Emission signals.

erikbosch, 2022-09-26:
This issue concerns VSS signals, so not necessarily connected to what is sent/available within Autosar/ARXML. The existing signal [Vehicle.EmissionsCO2](https://github.com/COVESA/vehicle_signal_specification/blob/c93ab10dabafe374d194fbd9278d9d32f4665f38/spec/Vehicle/Vehicle.vspec#L287) is an attribute, i.e. a fix value, so it is not even sure that the vehicle itself knows it, maybe known only be the backend.

Excerpt:

```
EmissionsCO2:
  datatype: int16
  type: attribute
  description: The CO2 emissions.
  unit: g/km
```

Anyhow, this attribute does not seem to be that well specified as it does not contain any information on how the value is or shall be calculated. There are some norms, like EU Regulation 2019/631 that possibly could be used. I do not know if e.g. US or China have other norms which might need to be supported as well.

I could see  a few ways forward here, if we see EU Regulation 2019/631 as the only relevant norm then we can just say that this attribute shall give the emissions according to EU Regulation 2019/631. If we can identify a few other norms that must be supported we could refactor so we would have separate signals for each norm, or come up with an array-solution where you can provide a list of norm/value tuples.

Then there is however the question if someone also needs to know instant emissions, as we do not have an signals for that either. If so - one need to discuss what is needed there, and it needs to be defined how the value shall be calculated, like how shall you calculate instant consumption in g/km when you are not moving.

If you have knowledge in what emissions signals that could be useful your contribution would be most welcome. Either just as an expert providing ideas and answering questions, or as a formal contributor preparing a pull request with possible improvements.

closed 2024-11-14 by erikbosch