COVESA/vehicle_signal_specification #395: Allow HasXXX in addition to IsXX for boolean

SebastianSchildt, 2022-01-11:
For some boolean sensors it might make more sense to name them "HasXX" instead of "IsXX", if we agree

 - change documentation to reflect that
 - change tooling to reflect that

erikbosch, 2022-01-18:
While discussing prefix for boolean signals, do we want to separate between singular and plural. For example, if a signal concerns status of multiple items, e.g. both front wipers, shall we then use:

- IsWiperWorn (not really correct as it concerns multiple wipers)
- IsWipersWorn (not really grammatically correct)
- AreWipersWorn (grammatically correct, but not that usual to use "Are" as prefix, right?)

danielwilms, 2022-02-01:
* `Is` instead of `Are`
* Find example for `has` and see it it's necessary

erikbosch, 2022-02-08:
Meeting notes:

- No need for "has" right now, might be added later if we have a good example
- Decision: close this ticket

closed 2022-02-08 by SebastianSchildt