COVESA/vehicle_signal_specification #396: How to document and where to place unit

danielwilms, 2022-01-11:
* how to keep docs and config in sync
* where to keep config and make it user friendly

slawr, 2022-01-11:
Related to this is to what extent the changes introduced with config.yaml have any rule set implications to document? One example would be the 'domain' key for each unit. Is this a defined list of allowed values or free-form?

slawr, 2022-01-11:
**Current sync state**
Table of unit types from the vss-tools config.yaml and the table in the VSS doc

|config.yaml|VSS doc|
|---|---|
|A|A|
|  Nm|Hz|
|  PS|N|
|  Pa|Nm|
|  UNIX Timestamp|Pa|
|  V|V|
|  celsius|W|
|  cm|celsius|
|  cm^3|cm/s^2|
|  day|degree|
|  degrees/s|degrees/s|
|  degrees|g|
|  g/km|g/s|
|  g/s|h|
|  h|inch|
|  inch|kPa|
|  kPa|kW|
|  kW|kWh|
|  kWh/100km|kg|
|  kWh|km|
|  kg|km/h|
|  km/h|l|
|  km|l/100km|
|  l/100km|l/h|
|  l/h|lat|
|  l|lon|
|  lbs|m|
|  m/s^2|m/s|
|  m|m/s^2|
|  min|mbar|
|  mm|min|
|  months|ml|
|  mpg|ml/100km|
|  percent|mm|
|  ratio|ms|
|  rpm|percent|
|  s|ratio|
|  stars|rpm|
|  weeks|s|
|  years||

danielwilms, 2022-02-01:
Decision to move config.yaml for specification of supported unit in to the vss repository being a required input for vss-tools.

closed 2022-12-14 by erikbosch