COVESA/vehicle_signal_specification #398: Is an attribute a signal?

erikbosch, 2022-01-14:
This repository/standard is called the COVESA Vehicle Signal Specification. But in the documentation we here and there states that only actuators and sensors are signals, not attributes. I see some problems with this definition - often when we talk about signals we actually mean all three entities (actuators, sensors, attributes). 

Would there be a problem changing the definition of signals so they also include attributes? Of course some documentation like the snippet below will need to be updated. As a side note, standards like VISS does not specifiy any difference between attributes and sensors - there are e.g. nothing that restricts you from subscribing to an attribute as I understand it.

Current description:

https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/

_Leaf nodes of the tree contain metadata describing the data associated to the node. In order to help application developers, who are using the specification, it makes a distinction between signals - in the following as sensor and actuator - and attribute. The difference between a signal and an attribute is that the signal has a publisher (or producer) that continuously updates the signal value while an attribute has a set value that should typically not change more than once per ignition cycle._

danielwilms, 2022-02-01:
Attributes and Signals are distinguished because of historical reasons. It should not be there anymore.

@erikbosch: will create a PR with a proposal.

jdacoello, 2022-03-01:
The concept of a `Signal` in VSS is misleading, and I think it should not be used as the reference to classify the leaves of the VSS tree. I see that the leaves of the VSS tree basically refer to a `property` of the vehicle that in practice will have a particular `value` that must comply with what is specified (e.g., Unit, min, max, allowed, etc.). 

From my point of view, I think it is easier to consider each leaf of the tree as a `property` and classify it as either..
* ..`Static` (i.e., an attribute that doesn't change), or 
* .. `Dynamic` (i.e., a property that has dynamic values)

With this consideration, the concept of a `Signal` would be mapped to `Dynamic` property, whereas an `attribute` will map to an `Static` property.

This is the first consideration we took for the mapping of VSS into VSSo. It might also be helpful fo the VSS specification itself.

PR #408 (2022-03-01): Changing documentation so attribute is considered a signal.
Previously we stated that only sensors and actuators were signals.

Fixes #398
commit dce58296 (2022-03-01): Changing documentation so attribute is considered a signal.

Previously we stated that only sensors and actuators were signals.

Fixes #398
closed 2022-03-01 by danielwilms