COVESA/vehicle_signal_specification #414: Support for high-order data types

paulboyes, 2022-02-17:
Support for high-order data types, to correlate rich sensor data with metadata. Example for a camera sensor, we need to model the image URL with the object list as part of the same sensor. Or is there a different way they recommend modeling this data.
struct Camera {
   String uri;
   List<String> objectList;
}
Vehicle.ADAS.Camera:
   type: sensor
   datatype: Camera
   description: “...”

erikbosch, 2022-02-18:
Structs as datatype is being discussed now and then, see e.g. #326

danielwilms, 2022-02-27:
I think the discussion went less in the direction of complex data types but rather into one realization problem. I would close this one as duplicate to #326.

closed 2024-11-14 by erikbosch