COVESA/vehicle_signal_specification #440: RPM inconsistency

SebastianSchildt, 2022-04-08:
Vehicle.Powertrain.CombustionEngine.Speed.   And Vehicle.Powertrain.ElectricMotor.Rpm

danielwilms, 2022-05-24:
@danielwilms create PR

UlfBj, 2022-05-25:
Before creating a PR I think we should step back a bit and think about the semantics of node names. 
Shall the semantics of the node name be context sensitive, or not? 
The context is in my view the path name from the root node and to the parent of the node in question. 
If not context sensitive, then e. g. "Speed" shall have the same meaning wherever it occurs in the tree, while if context sensitive its meaning depends on its ancestor nodes. 
My view is that it shall be context sensitive. 
This makes it possible to have the path names "Vehicle.Speed" and "Vehicle.Powertrain.CombustionEngine.Speed", where the semantics of Speed differs between the two. 
However, I think it is important to make this rule clear in the documentation, to minimise the risk of confusion that will occur if a reader believes we use a non context sensitive naming rule.

SebastianSchildt, 2022-05-25:
Agree. Definitely "context-sensitive" according to your definition.  just consider a potential signal Vehicle.Connectivity. Uplink.Speed ...

Also I guess there may  be many other situations, where problems arrive if we demand same leaf name means same "concept".  (in a sense rpm is also "movement per time unit", so it _might_  be considered a speed concept, but this finer semantics may be for the ontology colleagues to play with :) )

danielwilms, 2022-05-25:
I checked how we use it up to now we are pretty consistent with regard to the contextualisation.

This we use in OBD (< always the "problematic" one as it's an interface and all context is lost):
https://github.com/COVESA/vehicle_signal_specification/blob/53b4a3609b44001da0adc90f1a80731bf68a4c5d/spec/OBD/OBD.vspec#L123-L127

There we explicitly define the context.

In combustion engine, we use it, as you describe:
https://github.com/COVESA/vehicle_signal_specification/blob/53b4a3609b44001da0adc90f1a80731bf68a4c5d/spec/Powertrain/CombustionEngine.vspec#L131-L135

I checked the usage of `rpm` and we use it just once in electric motor, but refer everywhere in the documentation to `speed`.
https://github.com/COVESA/vehicle_signal_specification/blob/53b4a3609b44001da0adc90f1a80731bf68a4c5d/spec/Powertrain/ElectricMotor.vspec#L49-L53

I think changing `Rpm` to `Speed`, seems the right way to go forward.

closed 2022-05-31 by danielwilms