COVESA/vehicle_signal_specification #448: Exclude nodes from instantiation

danielwilms, 2022-05-22:
Currently, we instantiate all nodes under a branch, where instances are defined. As an example, we use:

```
Door:
  type: branch
  instances:
    - Row[1,2]
    - ["Left","Right"]
  description: All doors, including windows and switches.
#include SingleDoor.vspec Door
```

In this case all nodes in `SingleDoor.vspec` will be instantiated (e.g. `Door.Row1.Left.Status`). But there are nodes, which are directly related to the general concept of a  `Door` and not to an instance of a door (e.g. how many doors are in the vehicle). Now we solve this issue by adding a node on the parent node of `Door` and name it in a way, that the relation is clear:

```
# Default value based on instance declaration above with 2 rows and 2 doors in each row.
DoorCount:
  datatype: uint8
  type: attribute
  default: 4
  description: Number of doors in vehicle.
```

We thought, that it might be nicer to keep count as a value under the branch `Door` and give it an additional key to make clear that it's excluded from instantiation, like:

```
# Default value based on instance declaration above with 2 rows and 2 doors in each row.
Door.Count:
  datatype: uint8
  type: attribute
  default: 4
  instantiate: false
  description: Number of doors in vehicle.
```

That would keep the semantic structure and feels less like a workaround compared to the current solution. FYI @adobekan

erikbosch, 2022-05-24:
A possibly better example could be wheel/tire signals. We have today some signals on axle-level that logically belongs to wheel-level, but typically are the same for all wheels on the same axle, like TireWidth and WheelDiameter. With the feature proposed by we Daniel we could possibly move them to `Wheel`.

```
Vehicle.Chassis.Axle.Row1.TireWidth
Vehicle.Chassis.Axle.Row1.WheelDiameter
Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.Pressure
```

Could as well be ....

```
Vehicle.Chassis.Axle.Row1.Wheel.TireWidth
Vehicle.Chassis.Axle.Row1.Wheel.Diameter
Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.Pressure
```

... but it can be argued that we do not win that much from it.

I found however a possible alternative use-case for deployment/layering, related to the discussion in #447. Theoretically one could use an "instantiate" attribute to specify if a particular car e.g. report brake fluid level individually for each wheel or not. But then comes a corner case - what does `instantiate: false` means for signals with multiple parents with instantiations? Taking this example:

```
Brake.FluidLevel:
  datatype: uint8
  type: sensor
  unit: percent
  instantiate: false
  description: Brake fluid level as percent. 0 = Empty. 100 = Full.
```
Does that mean instantiation shall be removed for both axle and wheel

```
# Today
Vehicle.Chassis.Axle.Row1.Wheel.Left.Brake.FluidLevel
Vehicle.Chassis.Axle.Row1.Wheel.Right.Brake.FluidLevel
Vehicle.Chassis.Axle.Row2.Wheel.Left.Brake.FluidLevel
Vehicle.Chassis.Axle.Row2.Wheel.Right.Brake.FluidLevel

# Do we expect this (all instantiations removed):
Vehicle.Chassis.Axle.Wheel.FluidLevel

# Or this (only last instantiation removed):
Vehicle.Chassis.Axle.Row1.Wheel.FluidLevel
Vehicle.Chassis.Axle.Row2.Wheel.FluidLevel

```

danielwilms, 2022-05-24:
> what does instantiate: false means for signals with multiple parents with instantiations? Taking this example:

good point, I wasn't clear here. I assumed it as "overwriting" the instance attribute of the direct parent, as in your example:
```
# Or this (only last instantiation removed):
Vehicle.Chassis.Axle.Row1.Wheel.FluidLevel
Vehicle.Chassis.Axle.Row2.Wheel.FluidLevel
```

danielwilms, 2022-06-14:
@UlfBj, please have a look :)

UlfBj, 2022-06-15:
Following the rule mentioned by Daniel - "overwriting" the instance attribute of the direct parent - it seems ok to me.

SebastianSchildt, 2022-08-16:
Fixed.

closed 2022-08-16 by SebastianSchildt