COVESA/vehicle_signal_specification #457: JsonSchema Version of VSS

alexjoybc, 2022-06-16:
Hi this is a feature request

Adding [jsonschema](https://json-schema.org/) support for the specification

Why:

JsonSchema is machine readable and can be use for generating code, validating structure and is widely adopted by the community in general.

erikbosch, 2022-06-17:
I assume no one would object to adding a jsonschema, could be a good addition. I assume it needs to be used after converting *.vspec files to json as the VSS include handling for branches is not standard json.

But as usual - it relies on that someone has the time and interest to do it. We would be more than happy if you would have the time to work on it and create a pull request.

SebastianSchildt, 2022-06-17:
I tried looking into it in the past, but could not figure out (not a jsonschema expert), how to model the "meta" Aspects. It is relatively straightforward in jsonschema to say, "I want this JSON object to have an element foo which is required and string and an optional bar element that must be int", however for vspec we would need something along the lines of "every nested object needs to have a type", "If type equals sensor we can have...."

-> Not sure it is possible. If yuo know more about jsonschema @alexjoybc we could use some help/pointers here :)

alexjoybc, 2022-06-24:
I would like to help for sure, we are evaluating using vss for one of our project and having jsonschema available would be a good starting point for us.

tguild, 2022-06-29:
VSS' core truth is in YAML and both the current JSON and VSSo are derived through tooling. Those resulting formats have no immediate codependency so such a change should be fine. It is worth noting however that JSON-LD is more conducive towards ontologies while also retaining the benefits of JSON. On the other hand there seems to be more tooling, including code generators, and wider adoption of JSONSchema.

alexjoybc, 2022-07-01:
Yes I agree, jsonschema should be generated through tooling without impacting the current JSON definition and not impact the yaml core definition.
JsonSchema has greater adoption by the web community over JSON-LD, but again, following the guidelines of vss JsonSchema or JSON-LD should be supported over tooling, therefore one doesn't prevent the other and should be driven by community.

danielwilms, 2022-07-05:
- JSONSchema as additional exporter
- main use case openAPI, AsyncAPI, validation
- manual mapping up to now
- OpenAPI works with the tree, no need to make it fine granular 

How to move forward:
- @alexjoybc happy to contribute example/documentation

alexjoybc, 2022-07-06:
find bellow an example of using vss jsonschema with open api v3:

```json
{
    "openapi": "3.0.1",
    "info": {
        "title": "VSS with open api",
        "description": "",
        "version": "1.0.0"
    },
    "paths": {
        "/vehicle": {
            "post": {
                "summary": "Add telematics data for a vehicle",
                "requestBody": {
                    "description": "Vehicle telematics data",
                    "content": {
                        "application/json": {
                            "schema": {
                                "$ref": "#/components/schemas/Vehicle"
                            }
                        }
                    },
                    "required": true
                },
                "responses": {
                    "200": {
                        "description": "Success",
                        "content": {}
                    }
                },
                "x-codegen-request-body-name": "vehicle"
            }
        }
    },
    "components": {
        "schemas": {
            "Vehicle": {
                "type": "object",
                "properties": {
                    "Vehicle":
                    {
                        "$ref": "./vss_rel_2.2.json#/Vehicle" 
                        "description": "this is a reference to the Vehicle object described in the jsonSchema, could also be like ./vss_rel_2.2.json#/Vehicle/ADAS"
                    }
                }
            }
        }
    }
}
```

Then the VSS specification in jsonSchema

I made the following transformations without much thinking, just to get it started ;-)

* rename `children` to `property`
* rename `type` to `vss_type`
* rename `datatype` to `type`
* update types from

| VSS data type | JsonSchema data type |
| --- | --- |
| UInt8 | integer |
| Int8 | integer |
| UInt16 | integer |
| Int16 | integer |
| UInt32 | integer |
| Int32 | integer |
| UInt64 | integer |
| Int64 | integer |
| Boolean | boolean |
| Float | number |
| Double | number |
| String | string |

```json
{
    "Vehicle": {
        "properties": {
            "ADAS": {
                "properties": {
                    "ABS": {
                        "properties": {
                            "Error": {
                                "type": "boolean",
                                "description": "Indicates if ABS incurred an error condition. True = Error. False = No Error.",
                                "vss_type": "sensor",
                                "uuid": "cd2b0e86aa1f5021a9bb7f6bda1cbe0f"
                            },
                            "IsActive": {
                                "type": "boolean",
                                "description": "Indicates if ABS is enabled. True = Enabled. False = Disabled.",
                                "vss_type": "actuator",
                                "uuid": "433b7039199357178688197d6e264725"
                            },
                            "IsEngaged": {
                                "type": "boolean",
                                "description": "Indicates if ABS is currently regulating brake pressure. True = Engaged. False = Not Engaged.",
                                "vss_type": "sensor",
                                "uuid": "6dd21979a2225e31940dc2ece1aa9a04"
                            }
                        },
                        "description": "Antilock Braking System signals",
                        "vss_type": "branch",
                        "uuid": "219270ef27c4531f874bbda63743b330"
                    },
...
```

gunnarx, 2022-09-22:
I am not convinced that "converting" VSS definition to JSON schema really makes sense.  They don't seem to serve the same purpose in some sense (see last comment).   At least not without adding some more details to the intended outcome and what is actually desired here?

> without impacting the current JSON definition

Please explain first what you mean by _the current JSON definition_, so we're not missing something.

> I made the following transformations without much thinking, just to get it started ;-)

Unless I am misunderstanding you I see something that I think is a common mistake/pattern here and that is to mix up the _definition of signals_ with the definition of a value-transfer protocol or API for those signals.  When you perform the POST operation defined in the API, and you send a document that follows the format described in the JSON schema, how do you transfer values of the signals, or is that not what you are trying to do?  (Hint:   there is no property named "value" in what you have described)

adobekan, 2022-11-24:
Here is one more nice example on how to use jsonSchema. Might help with clarification.
https://www.asyncapi.com/blog/json-schema-beyond-validation

gunnarx, 2023-02-20:
> Here is one more nice example on how to use jsonSchema. Might help with clarification.

That depends what you want to clarify and to whom.  I know what JSON schema does - it describes the structure of a JSON document, and tools can check if a JSON document is correctly formatted, according to the schema.

The question in the discussion above is another: 
"What kind of document is it that you you want to check?"

adobekan, 2023-02-20:
> I know what JSON schema does - it describes the structure of a JSON document, and tools can check if a JSON document is correctly formatted, according to the schema.

Not that I was trying to refer to something in your comment. But I am happy for you that you know what for JSON schema is used. My comment was going more into the direction of one more general example of VSS and JSON Schema. 

> "What kind of document is it that you you want to check?"

At the end, if you use JSON to serialize data in this case VSS data, then you can use json-schema to validate your API req/response. Yes you are right it could be for ViSSv2 or any other API that is handling in this case VSS-data. 

Cheers

tguild, 2023-08-02:
@alexjoybc there is a pending PR for this in related repo

https://github.com/COVESA/vss-tools/pull/296

alanmapleburl-au, 2023-09-07:
I'll add my two cents.  I think it would be beneficial to have the source of truth in JsonSchema or OpenAPI with a tool to generate legacy vspec for backward compatibility.  They're both more expressive than vspec and are well-known.