COVESA/vehicle_signal_specification #458: Possible duplicated concept (Gear)

jdacoello, 2022-06-21:
Apart from the fact that one is specified as `Sensor` and the other as an `Actuator`, is there any difference between `CurrentGear` and `SelectedGear`?

I think they can be merged into one. 

The specification has the following:

### CurrentGear:
  datatype: int8
  type: sensor
  description: The current gear. 0=Neutral, 1/2/..=Forward, -1/-2/..=Reverse.

### SelectedGear:
  datatype: int8
  type: actuator
  description: The selected gear. 0=Neutral, 1/2/..=Forward, -1/-2/..=Reverse, 126=Park, 127=Drive.

erikbosch, 2022-06-21:
The gear signals were discussed (#328) and modified quite recently. For a vehicle with manual transmission there should normally be no difference. But for an automatic vehicle they are typically different. There `SelectedGear` is often `127==Drive` while `CurrentGear`represents the mechanical gear used, i.e. typically a value between 0 and X where X depends on number of gears in the automatic gear box.  

(As a side comment - Android also has two signals for gears, see
https://android.googlesource.com/platform/hardware/interfaces/+/master/automotive/vehicle/2.0/types.hal#666)

danielwilms, 2022-06-21:
intentional, documentation update could help.