COVESA/vehicle_signal_specification #459: Multiplicity of Electrical Engines - how to represent in VSS?

erikbosch, 2022-06-27:
In VSS we have today a branch `Vehicle.Powertrain.ElectricMotor` for information on electric motor(s) in the vehicle:

```
#
# The electric motor branch used to host the signals and attributes in ElectricMotor.vspec.
#
Vehicle.Powertrain.ElectricMotor:
  type: branch
  description: Electric Motor specific data.

# Include the electric motor vspec file and attach all its signals and attributes under the
# electric motor branch created above.
#include Powertrain/ElectricMotor.vspec Vehicle.Powertrain.ElectricMotor
```

Vehicles with multiple electrical engines are becoming more usual, e.g. one engine for front axle and one engine for rear axle. Getting individual information for each engine is not possible with current VSS, and this is a possible improvement area for VSS.

But how how would that best be handled? Adding just instance information like below (to support a vehicle with separate engines for each wheel) is possibly, but would likely not fit all vehicles. There could e.g. be 4-wheel-drive electrical cars using a single engine and a traditional 4wd transmission.

```
Vehicle.Powertrain.ElectricMotor:
  type: branch
  description: Electric Motor specific data.
  instances:
    - ["Front","Rear"]
    - ["Left","Right"]
```
Is this possibly a use case for overlays/extensions. E.g. to keeping current (single engine) specification, and adding an overlay adding the part below to change to use two engines instead of one? Or do we see a better solution to support both vehicles with a single engine and vehicles with multiple engines?

```
  instances:
    - ["Front","Rear"]
```

danielwilms, 2022-06-28:
Discussion:
- maybe remove all instances from standard catalog and add standard profile overlay

CorvetteCole, 2023-01-25:
I think this seems like a place for overlays, but I agree it is an increasingly important feature (one needed for a car I am working on)

erikbosch, 2024-11-14:
We regularly close old issues and pull requests that appear to be abandoned or completed, where there have not been any activity for a long time. If you think your item is still relevant please feel to add a comment and re-open it. You are also very welcome to bring up the topic at one of our weekly VSS meetings.

yf6, 2025-07-29:
I'd like to reopen this issue for consideration, it would be great to be able to represent multiple electric motors.

erikbosch, 2025-07-30:
Reopened on request by @yf6