COVESA/vehicle_signal_specification #466: Propose to extend VSS to include cybersecurity properties and impact ratings

selfighter, 2022-07-13:
Cybersecurity is becoming increasingly important as the auto industry is advancing in the direction of connectivity、autonomous、and smart, because more impact can be caused by cyber attacks now. Now we got automotive cybersecurity regulations(eg. UN Regulation No. 155 - Cyber security and cyber security management system ) and standards ( eg.ISO/SAE 21434:2021 Road vehicles — Cybersecurity engineering) ,so it is of value to extend VSS to have cybersecurity support.
There are initiatives like ThreatCatalog by ASRG(Automotive Security Research Group),with is trying to catalog the cyberthreats and associated risk without giving enough  detailed automotive domain specific information about impacts .
So I think as we already have VSS as a catalog of typical signal in automotives，if we extend it to include cyber threats ，its associated security property（eg. CIA ,Confidentiality，Integrity，Authenticity）and its impact of its cybersecurity property been broken. It can answer the questions ,for example ,"what is the impact if the speed signal's integrity were broken thus  been spoofed   ?" The answer could be like"If the speed is been spoofed can result in sudden acceleration、stoping etc. and causing traffic accidents to threat driver's life.", with those answers combined with threat and attack feasibility  in mind ,security architect can then come up with threat mitigation and controls (eg. adding MAC ,Message Authentication Code )to prevent this kind of risk.
VSS has automotive specific data catalog, we can extend data/parameters in VSS's to include cybersecurity properties like potential impact（SFOP ） of breaking certain cybersecurity properties（eg, CIA） and threats（eg. STRIDE  ）.
I am attacking a example here modified from VSS files, for more details pls refer to aforementioned regulation and standard:

  CruiseControl.SpeedSet:
  datatype: float
  type: actuator
  unit: km/h
  description: Set cruise control speed in kilometers per hour. 
  confidentiality: not required
    securityImpact: negligible
  integrity: required
    securityimpact: severe,could allow hackers to spoof the cruse speed .
  authenticity: required
    securityimpact: xxxx

SebastianSchildt, 2022-07-13:
Hi @selfighter  this would be a good example for an overlay with arbitrary attributes. Check the documentation here https://covesa.github.io/vehicle_signal_specification/rule_set/overlay/
Only the most recent (master) tools have support for this, see "Handling of overlays and extensions" in https://github.com/COVESA/vss-tools/blob/master/docs/vspec2x.md

I think what VSS *can not* do is define waht exactly are the values of "integrity", "securityimpact" etc. for all data points, as I assume that is als depenedent on use case/implementation. 

What we maybe *should* do is, defining which keywords are needed, following one/more of the activities you mentioned and also describing the possible values. Basically what you started as an example, with a little bit more explanation around, and then some example overlay for a few datapoints that can be ingested by tooling (just to show how it would be done).

As obvously, you already know more about this, this group would be very happy, if you can help us there 👍

erikbosch, 2022-08-02:
Meeting notes: would be good if it could be based on some standard. If more are interested and have time to drive it could be a possible addition/recommendation for VSS

closed 2024-11-14 by erikbosch