COVESA/vehicle_signal_specification #473: Intended content of a release

erikbosch, 2022-08-05:
The [latest release](https://github.com/COVESA/vehicle_signal_specification/releases/tag/v3.0) contain the following assets

- [vss_rel_3.0.json](https://github.com/COVESA/vehicle_signal_specification/releases/download/v3.0/vss_rel_3.0.json)
- [vss_rel_3.0.tar.gz](https://github.com/COVESA/vehicle_signal_specification/releases/download/v3.0/vss_rel_3.0.tar.gz)
- Source code (zip) 
- Source code (tar.gz) 

The first two are manually added. That tar.gz file contain output from all "supported" tools in vss-tools except binary, as binary output might be target specific). The json file (generated by vss-tools json exporter) exist in the tar.gz file as well but is also added separately to simplify showing VSS in online JSON tools.

@slawr has identified in a slack discussion that the content may be confusing and might need to be changed or better explained.

slawr, 2022-08-05:
### Input on Release Assets
I have a feeling I may have said this before, so apologies if I have and we discussed it..

Personally, I think the 'assets' of the release are presented a little strangely for people not very familiar with the project. Especially for newcomers to understand.

Looking at it from that perspective:
- The yaml is the single source of truth for the standard catalog but if I open v2.2.tar.gz I don't see the yaml, but generated serialisations with no doc or comment inside them to tell me that. Of course the newbie will not know this.
- The assets also includes vss_rel_2.2.json in the root and I might wonder is this .json the catalog, which is confusing as the doc talks about .yaml.. or maybe I'm just moving very fast and assume that the json is the source of truth. Side note: I see Daniel's explanation on slack that the json is there to drive the json viewer embedded in the doc.
- If I finally open the source code archive then I may then find a yaml file if I hunt around.

After a while you may figure out that the source code archive contains the catalog and that the v2.2.tar.gz contains generated output from vsc-tools, etc. but its not obvious early on.

How to improve that is a matter of details, but could be easily fixed in various straight forward ways. You could have a few lines in a readme.md/release.md released as an asset, which summarises what the files are using vX.Y.tag.gz filenames that don't need to be maintained. Another would be more descriptive archive names.

erikbosch, 2022-08-09:
Meeting comments:

- Shall individual files be added separately? (i.e. not in *.tar.gz)

slawr, 2022-08-30:
### Deprecation
Should the review consider to what extent we document deprecation in the rule set or standard catalog?

Currently the release note has a "What's Changed" section that includes a summary line for all changes but finding changes is largely left to an end user exercise in vspec comparisons between releases.

erikbosch, 2022-08-31:
So far for our releases we typically summarize the changes in a few lines and then also include an auto-generated release note ("What's Changed") with the headers of all commits. We do not have a detailed list showing exact changes in the vspec files.

But the question is - what changes does someone actually want to find or be informed of? It would be feasible to create and include (manually or automatically) a summary list of added/removed/deprecated signals, but do we also want to include changes of description, datatype and comment?  

Concerning deprecation we have some guidelines in https://covesa.github.io/vehicle_signal_specification/rule_set/basics/#deprecation-since-version-21. So far it has worked quite well when removing/renaming a signal - if next planned release is a minor release we duplicate the signal and mark the old name as deprecated. If next planned release is a major release we just rename it. For other changes the deprecation keyword does not work, like if you change datatype, unit, description of instantiation, or add/remove properties. We cannot e.g. at the same time have one signal "Speed" with int16 as datatype and one with float as datatype.

closed 2024-11-14 by erikbosch