COVESA/vehicle_signal_specification #479: UUID in release artifacts

erikbosch, 2022-08-22:
As of default vss-tools by default include UUIDs, and thus release artifacts by default include UUIDs. 

Two thoughts:

- Is anyone actually using the UUIDs?
- Would it be an idea to change default so that UUID only is included if requested, implying that releases will be without UUID

adobekan, 2022-08-23:
Would work for us to have optional build tag for UUID.

erikbosch, 2022-09-06:
See meeting notes in https://github.com/COVESA/vss-tools/pull/189:

At least for minor releases (in 3.X) we should not remove UUID, but rather possibly generate both, and remove the old variant with UUID first in next major release (v4.0)

erikbosch, 2022-12-21:
@adobekan - Found this issue, if you intend to refactor the UUID so it actually is useful then it might be an idea to keep UUID in released artifacts.

closed 2024-11-14 by erikbosch