COVESA/vehicle_signal_specification #482: driver.vspec feels a little under specified

slawr, 2022-08-23:
### Micro view
https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/Driver/Driver.vspec contains various leaves that read as being related to Driver Monitoring Systems (DMS). For example fatigue or alertness states.

However the vspec does not contain sufficient information in comments or the description to be able to interpret the values.

For example:

```
AttentiveProbability:
  datatype: float
  type: sensor
  unit: percent
  min: 0
  max: 100
  description: Probability of attentiveness of the driver.
```

At a minimum there is no explanation of what the extremes of the values represent. Does 0 represent fully attentive or asleep for example?

This is the case for all of the leaves using percents, i.e. FatigueLevel, AttentiveProbability and DistractionLevel.

Not knowing the background to the additions I can not easily add the missing text hence the issue raised here.

### Macro view
A related higher level question is whether two years on whether the fundamentals of the leaves still fit production needs. 
As an example visual parts of Driver Monitoring Systems often monitor eye open state and point of focus (e.g. phone, windscreen, dashboard) but I wonder how they would map to these leaves.

This would ideally get input from DMS vendors.

slawr, 2022-09-06:
Discussed in the VSS call today. I will send a PR fixing the description gap for the existing nodes.

adobekan, 2022-09-07:
Note: focus would be on improving description of the existing VSS nodes. Thanks @slawr.

closed 2024-11-14 by erikbosch