COVESA/vehicle_signal_specification #484: Logo Formats

paulboyes, 2022-08-24:
We should provide other formats beside svg and/or build them with releases?  I made some today for COVESA Marketing at their request.  

[image]
[image]
[image]
[image]

SebastianSchildt, 2022-08-25:
I feel for the repo, we should only have "the source" (e.g. SVG) , that allows easy generation of everthing else , _unless_ we want to put more variants into the documentation here: I assume, you are thinking about something like a "marketing" page which provides logos in pdf, (e)ps, different PNG solutions etc.? That could either live  in our VSS documentation or in COVESA Wiki (somewhere around the "landing" page)

erikbosch, 2022-08-30:
Meeting notes: For now only keep sources in git, building/generating artifacts shall be done manually, i.e. not included in git/github Makefile/CI/Release-process

closed 2022-08-30 by erikbosch