COVESA/vehicle_signal_specification #493: COVESA Contribution guidelines

erikbosch, 2022-09-20:
Took a look at the [COVESA Contribution guidelines](https://www.covesa.global/contribute) and found some topics that we have not been following strictly:

*Every contribution must carry the following sign-off line with your real name and email address:
"Signed-off-by: Firstname Lastname <[firstname.lastname@example.com](mailto:firstname.lastname@example.com)>"*

*When you submit code for the first time, the project maintainer will contact you and make sure that you are familiar with the DCO and answer all the questions you might have regarding contributions.*

I think it could be good to briefly discuss this - if we should start enforcing it. And how to manage if someone does not have an e-mail address of the format `firstname.lastname@mycompany.com`.

erikbosch, 2022-09-20:
Meeting notes:

- COVESA Guidelines are up to date and shall be followed
- Paul with check within COVESA if repos can be setup to enforce the check
- Regardless, we should as part of reviews verify that there is a signoff before merging

adobekan, 2022-10-04:
Note 04.09.22
@paulboyes will check if we can enable this check by default.

erikbosch, 2022-10-11:
Now described in #491. But we shall maybe leave it open until we have a conclusion on the automatic check

closed 2022-10-11 by erikbosch