COVESA/vehicle_signal_specification #502: Support for int/float/double together with "allowed"?

erikbosch, 2022-10-25:
As of today we use "allowed" only for string in standard VSS catalog, but using it for other datatypes is not forbidden, see [documentation](https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/allowed/).

However, some of our tools (like vss2ddsidl) does not seem to be able to handle float or int, giving errors like:

`TypeError: sequence item 0: expected str instance, float found`

In the past we have said that supporting "allowed" for any datatype makes sense, and if so this is a pure bug/limitation for those tools. Supporting int-types makes sense as I see it, but do we see it as relevant to support float/double, as checking equality for floating point numbers could be somewhat complex.

erikbosch, 2022-10-25:
Meeting notes:
Ulf thinks it can shall be limited to strings. Float is problematic, we shall at least not allow float/double.
Erik thinks supporting it on integers could make sense, if we want to only support it for string we might impact anyone using it today (for unofficial signals), as most tool support int (and float) today.

Way forward: Discussion to be continued in coming meetings.

adobekan, 2022-11-08:
Follow up once @erikbosch is back from vacation. Q: Where would we need this? Any UC in mind.

erikbosch, 2022-11-14:
As of now no known use-case for VSS standard catalog. But if we want to separate VSS catalog from VSS-syntax and VSS-tools (so that syntax and VSS-tools theoretically can be used for other models) then I see two options:

- We can change documentation (and tooling) so that an error is given if trying to use "allowed" on something else than strings
- We keep it as supported for all types in the syntax, but then we should list the limitation and/or give an error message if used for those tools that does not support it.

kkoppolu1, 2022-11-15:
A use case for supporting allowed values for integral types (integers) is enumerations. For example, CAN-DBCs contain VAL_TABLE_s for int to string mappings.

erikbosch, 2022-11-15:
Meeting notes:
Krishna described potential use cases for integers
Daniel Alvarez prefers that allowed only if supported for strings
Discussion to be continued in following meetings

erikbosch, 2022-11-29:
Meeting notes:

- Sebastian: Does not see why limited to strings, at least also integer, possibly also float
- Sebastian: Lets write down options and let us vote in issue.

erikbosch, 2022-11-29:
Until next meeting (2022-12-06) please vote by adding a comment using the following template. Support of `allowed` for strings will be kept regardless of result

- Shall VSS syntax support `allowed` for integer-based types: [YES|NO]
- Shall VSS syntax support `allowed` for float-based types: [YES|NO]
[Your name and which COVESA member organization you are representing]

erikbosch, 2022-11-30:
I think 'allowed' could be useful for both int and float types in some deployments, for example as a mechanism to communicate supported values for e.g. wanted temperature in an overlay, e.g. '18.5, 19.0, 19.5, ..., 23.0'. For the float comparison problem I consider that it is up to the implementation to decide on how to compare, if it e.g. consider a set request for 18.47 to be a valid request for 18.5 (closest allowed value) or if it consider it to be an illegal request (too much difference compared to 18.5).

So my vote is:  

    Shall VSS syntax support allowed for integer-based types: **YES**
    Shall VSS syntax support allowed for float-based types: **YES**
    Erik Jägervall, Bosch

SebastianSchildt, 2022-11-30:
I think for int this is really useful, and I would consider not supporting it a "bug". For float, for discussed reasons you need to be careful and  for some use cases it might not be recommended, but arbitrarily forbidding it seems unexpected behaviour to me

Therefore

```
Shall VSS syntax support allowed for integer-based types: **YES**
Shall VSS syntax support allowed for float-based types: **YES**
Sebastian Schildt, Bosch
```

kkoppolu1, 2022-12-13:
I am aligned on supporting allowed values for integers and strings.
I am a bit unsure about supporting allowed values for floating-point nodes. It seems like we are passing the burden of floating point validation nuances to the implementations. But this is not a blocker for me. If the spec goes this way, I am OK with it.

erikbosch, 2022-12-13:
vote result: allow int and float for now

closed 2023-01-24 by erikbosch