COVESA/vehicle_signal_specification #505: Definition of "current trip"

erikbosch, 2022-11-01:
We have some signals related to "current trip" in VSS, but they do not really give any definition on what is meant with "current trip". Do we consider "current trip" to be something that is manually controlled by the driver (or a vehicle app, like automatic reset when a new rental period begins), i.e. there is no definition that say that starting the engine means that a new trip automatically begins?

A side topic - how does OEMs typically calculate average speed for "current trip". Do they consider all time when "Vehicle.LowVoltageSystemState" is on, i.e. considering time when engine is idling? Or do they exclude time when transmission is in neutral? 

```

ConsumptionSinceStart:
  datatype: float
  type: sensor
  unit: l
  description: Fuel amount in liters consumed since start of current trip.

TimeSinceStart:
  datatype: uint32
  type: sensor
  unit: s
  description: Time in seconds elapsed since start of current trip.

TripMeterReading:
  datatype: float
  type: sensor
  unit: km
  description: Current trip meter reading.

AverageSpeed:
  datatype: float
  type: sensor
  unit: km/h
  description: Average speed for the current trip.

```

adobekan, 2022-11-08:
@jdacoello will have a look for BMW details.

erikbosch, 2022-11-15:
Meeting notes:

Daniel Alvarez - current trip is from "start of car" to "stop of car". (from pressing "start button" to pressing "stop button"). It shall better be described how calculation was done for AverageSpeed. E.g. only considering when vehicle is moving?
Erik - But is Trip meter referring to the trip meter that can be manually reset.  E.g. current value of trip meter, or value of current trip.
Erik - For AverageSpeed, can we in VSS define how average speed is calculated, or say that it is OEM specific
Erik to create PR for further discussion

erikbosch, 2023-02-07:
Can be closed as PR merged

closed 2023-02-07 by erikbosch