COVESA/vehicle_signal_specification #508: VSS release schedule and features roadmap

erikbosch, 2022-11-17:


erikbosch, 2022-11-18:
Some ideas for discussion on next meeting:

- A minor release (v3.1) around Christmas/NewYear. Nothing really new except what we already have merged since v3.0, so no reason to relese it as 4.0
- A major release (v4.0) - to be released at latest in June 2023, possibly already before AMM, intended to include some major features like: Struct support (#326) and move of unit declarations to VSS (see #396 #346)

erikbosch, 2022-11-22:
Meeting comments:

- Krishna to think about release schedule/process, do we instead want a fixed schedule, do we need to separate between breaking/non-breaking changes
- Stephen: Roadmap should contain defining release process/cadence
- Krishna: Some changes that may affect backward compatibility, e.g. "allowed" not allowed for float-numbers.

SebastianSchildt, 2022-11-29:
I would agree that we could do a 3.1 "anytime" with just the updates to the standard catalogue so far

erikbosch, 2023-10-26:
_Erik 2023-10-26: Considered done for now, closing_

closed 2023-10-26 by erikbosch