COVESA/vehicle_signal_specification #510: Add Display Brightness Level for Vehicle Power Optimization

UmangSharmaMobis, 2022-11-21:
It need to have a Display Brightness Level for Vehicle Power Optimization

UmangSharmaMobis, 2022-11-22:
The Display on/off status with power efficiency parameter signal makes OEMs to analyze the optimization algorithms and select the suitable display unit.

UmangSharmaMobis, 2022-11-22:
The Display on/off status with power efficiency parameter signal makes OEMs to analyze the optimization algorithms and select the suitable display unit.

Notes:
Proposed that VSS Signals can be added in two different ways for this project: 

1. Add in the Existing VSS Modules in the Infotainment Category. 
2. Add a separate VSS Module for EV Power Optimization.

UmangSharmaMobis, 2022-11-22:
The Display on/off status with power efficiency parameter signal makes OEMs to analyze the optimization algorithms and select the suitable display unit.

Notes:
Proposed VSS Signals can be added in two different ways:

1. Add in the Existing VSS Modules in the Infotainment Category.
2. Add a separate VSS Module for EV Power Optimization.

based on the VSS meeting members feedback approach will be decide

UmangSharmaMobis, 2022-11-22:
[image]

UmangSharmaMobis, 2022-11-22:
The Display on/off status with power efficiency parameter signal makes OEMs to analyze the optimization algorithms and select the suitable display unit.

Notes:
Proposed VSS Signals can be added in two different ways:

Add in the Existing VSS Modules in the Infotainment Category.
Add a separate VSS Module for EV Power Optimization.
based on the VSS meeting members feedback approach will be decide

[image]

UmangSharmaMobis, 2022-11-22:
The Display on/off status with power efficiency parameter signal makes OEMs to analyze the optimization algorithms and select the suitable display unit.

Notes:
Proposed VSS Signals can be added in two different ways:

1. Add in the Existing VSS Modules in the Infotainment Category.
2. Add a separate VSS Module for EV Power Optimization.

based on the VSS meeting members feedback approach will be decide
[image]

erikbosch, 2022-11-22:
I am working on a comparison betwen VSS and [VHAL](https://android.googlesource.com/platform/hardware/interfaces/+/master/automotive/vehicle/2.0/types.hal) and there we have identified a gap, Android has `DISPLAY_BRIGHTNESS`, VSS does not have anything similar. As this issue came up I created a PR with 2 HMI-related changes, as it can be discussed as a possible solution.

What needs to be discussed as I see it:

- The PR #512 propose a signal with relative brightness (percent, from minimum supported/allowed to maximum supported/allowed), would this be sufficient for the power optimization use case
- Do we actually need a separate branch for "PowerOptimize" like in the comment above? A signal `Vehicle.Cabin.Infotainment.HMI.Brightness` could possibly suffice, and whether it is used for power optimization or does not then matter. We have so far in VSS-history avoided to have duplicate signals in different branches containing same information. Before adding a new branch `PowerOptimization` for this particular signal we would need to discuss power optimization in general - What signals are needed that do not fit in already existing branches?

UmangSharmaMobis, 2022-11-22:
their are few signals needed in case of power optimization. when drive turn on power optimization. 
system DisplayBrightnessLevel will behave in optimize manner based on algorithm.

and when user turn off power optimization in that case system should goes to previous state.

UmangSharmaMobis, 2022-11-22:
[image]

based on approach signal can define.

Signal Sample Code:

PowerOptimize:
type: branch
description: All PowerOptimize Actions

PowerOptimize.DisplayBrightnessLevel
  datatype: uint8
  type: actuator
  min: 0
  max: 100
  unit: percent  
  description: Power Optimize Display Brightness Level

erikbosch, 2022-11-22:
Meeting notes:

- Umang and Pradeep describe use case. Proprietary algorithm. If some mode is active, then certain signals are set.
- Adnan: Data can generally be added to existing structure.
- Erik presented a proposal - keep "normal signals" in "normal tree" but add specific signals to control power optimization in separate directory, possibly like:
```

Vehicle.PowerOptimize.Level:
  datatype: uint8
  unit: percent
  type: actuator
  description: Level of optimization, 0 = No savings, 100 = Most aggresive 
  
Vehicle.PowerOptimize.IsEnginePowerOptimized:
  datatype: boolean
  type: actuator
  description: Is Power optimization in engine is enabled 
```

... then PowerOptimized can use that data to control e.g. brightness.

- Umang and Pradeep to think about what works best and present an update in a coming meeting.

erikbosch, 2022-12-15:
Reopening it, more signals may be needed

UmangSharmaMobis, 2022-12-20:
added Power optimization signal in https://github.com/COVESA/vehicle_signal_specification/pull/520 pull request.

erikbosch, 2023-09-18:
I think this one can be closed now

closed 2023-09-18 by erikbosch