COVESA/vehicle_signal_specification #513: Release Cadence

erikbosch, 2022-11-23:
I create this issue to track the discussion that was originated as part of #508 

**Background**

As of today VSS has no fixed release schedule. Historically almost all development takes place in master-branch, and a release has been created about every 6 months, typically based on whether there is enough changes to justify a release or not. (There are some overhead to create a release and historically our CI-testing have not been that good, so some testing has been required.) If master contains "significant changes" like major extensions to syntax or backward incompatible changes (likely to affect someone) then it has been released as a major release (X.0), otherwise as a minor release (X.Y+1). The "official" documentation at https://covesa.github.io/vehicle_signal_specification/ always correspond to "master", we do not as of today store

**Possible Changes**

It is relatively easy to agree on a fixed schedule, it becomes a bit more challenging if we are to work on several releases in parallel, i.e. stop developing only in master, but rather have several active branches. We have a branch `release/3.0` created for last release, so it would be possibly to use that branch (or a branch `release/3.1`) for non-breaking/non-significant changes and continue to use `master` for items intended first for next major release (and merge `release/3.1` to `master` at regular intervals).

This would imply that "minor" changes, like adding of a signal, shall be primarily directed to the "minor"-branch (e.g. `release/3.1`) while significant changes shall be directed to `master`branch. We would also need to discuss for how long time we shall add "minor" changes to a released version.

Example:

If we add a signal A.B today it could be a candidate for both v3.1 and v4.0. But shall we stop adding signals to v3.x after we have released v4.0? Or have a more time-based approach, like "we will continue to add new signals to v3.x as long as less than 3 months have passed since v4.0 was released or less than 12 months has passed after v3.0 has been released."
 
**References**

- https://github.com/COVESA/vehicle_signal_specification/blob/master/docs-gen/content/governance/_index.md
- https://github.com/COVESA/vehicle_signal_specification/wiki/Release-Instructions-and-Checklist
- https://github.com/COVESA/vehicle_signal_specification/blob/master/RELEASE_PROCESS.md

SebastianSchildt, 2022-11-25:
For practical/manpower reasons I would try not to support/develop too many releases in parallel. I think in general having a cadence, e.g. every 3-4 months would be good, and relatively easy for standard catalogue changes (e.g. new/modified signals). That would give some more predictability when a merged signal will be "officially in".

For things warranting major releases (I think they can not come on a fixed cadence, but we should still have a roadmap for the next, and maybe next next major one feature wise), it is a little harder. E.g. taking struct support as an example: Likely you want it in a major release, and do not want any half-done, potentially breaking changes in the point releases before, so there may be window where two branches need to be considered. But in this specific example it is still not that hard, there are 3 things to consider

 - tooling may slowly learn to enable the upcoming "Major release" features, but we can always tie an older version to a release
 - Standard catalogue probably not an issue at all (becasue before having the struct feature defvined and implemented we will not begin adding such signals
 - Documentation: This may be harder, because I think once we merged something to master it is live? Maybe here we need to change the process, e.g. having multiple (latest release and "nightly") versions of the webpage, or as a simple hack - just in text mark things that are WIP and not released yet

erikbosch, 2024-02-12:
Some has been documented at https://github.com/COVESA/vehicle_signal_specification/wiki/VSS-Version-Governance-(Major-Minor). I think this one can be closed

closed 2024-02-12 by erikbosch